SpikeInterface/spikeinterface #1827: Export sorted units from spike interface

claudio1212, 2023-07-16:
I have a question regarding the use of spikeinterface that I know could seem a little "obvious": I started to use this tool just a few days ago, trying to run spike sorting with various sorters, starting from an open ephys recording stored in an .mda file. I privilege, for the moment, mountainsort5, with which I managed to run sorting through spike interface. I'd like to export each sorted unit, with the associated spiking events/times and the channel ids indicating where the sorted units were recorded, possibly in a .mat file. At the moment, I don't know how to do this and I would be really grateful if you could help me!

alejoe91, 2023-07-18:
Hi @claudio1212 

We have recently discussed about it and we plan to make an export function for sorting objects to .mat format. Would that solve your issue?

claudio1212, 2023-07-18:
Yes, it would definitely be an upgrade for my current pipeline that relies on Matlab for many analyses. Thank you really much for the response!

samuelgarcia, 2023-07-18:
Why not using NWB for exporting and the nwb matlab reader ?

claudio1212, 2023-07-18:
https://neurodatawithoutborders.github.io/matnwb/ are you referring to this pacakge? If so, I am surely going to try it. Currently, I  may have found a possible solution using the following code, but I am not sure it is the right thing to do if you want to export the information I mentioned in my question:

import mountainsort5 as ms5

sorting_MS5 = ms5.sorting_scheme2(recording=recording_preprocessed_MS5,sorting_parameters=ms5.Scheme2SortingParameters(detect_sign=1,detect_time_radius_msec=1.5,phase1_detect_channel_radius=2,detect_channel_radius=1))

folder = 'waveforms'
channel_ids = sorting_MS5.get_unit_property(unit_id,'channel_id')
file_path=r'D:\Utenti\Laboratorio\Desktop\on_the_fly\spike_interface\sorting\mysort.npz'

folder = 'waveform_folder'
we=si.extract_waveforms(
    recording_preprocessed,
    sorting_MS5,
    folder,
    max_spikes_per_unit=None,
    overwrite=True
)

####extract index of channels
chdict=si.core.get_template_extremum_channel(we, outputs='index')
unit_ids = sorting_MS5.get_unit_ids()

spike_times=[]
channel_ids =[]
for unit_id in unit_ids:
  spike_times.append( sorting2.get_unit_spike_train(unit_id)/30000)
  channel_ids.append(chdict[unit_id]+1)

# Save the MATLAB data

import scipy.io as sio    
sio.savemat(r'D:\Utenti\Laboratorio\Desktop\on_the_fly\spike_interface\sorting\results.mat', {'spike_times': spike_times,'channel_ids': channel_ids})

h-mayorquin, 2023-07-19:
I also think it is better to export to some intermediate format instead of matlab.  You have also have the scipy solution as you mention and you can use if you want to save in the new matlab format:

https://github.com/frejanordsiek/hdf5storage

zm711, 2024-04-30:
Is this still planned (to export to `.mat`) or should we close this?

zm711, 2024-06-20:
Let's close this for inactivity I think this issue has listed a variety of suitable solutions.

closed 2024-06-20 by zm711