SpikeInterface/spikeinterface #1828: Problems with SC2 and new SI for sliced recordings

yger, 2023-07-16:
I don't think I'm doing something wrong, but with 0.98, something is wrong with SC2. The following code for example

import spikeinterface.full as si
rec = si.read_spikeglx('NP1/')
rec_sliced = rec.frame_slice(0, 2*30000)
print(rec_sliced)
res = si.run_sorter('spykingcircus2', rec_sliced, verbose=True, output_folder="sorting")

leads to a crash. I have the proper display of the recording (that can be displayed, ...):

FrameSliceRecording: 384 channels - 30.0kHz - 1 segments - 60,000 samples - 2.00s - int16 dtype 
                     43.95 MiB

But when launching SC2, I have the following crash

 File "/home/pierre/github/spikeinterface/src/spikeinterface/core/base.py", line 987, in _load_extractor_from_dict
    extractor = extractor_class(**new_kwargs)
  File "/home/pierre/github/spikeinterface/src/spikeinterface/extractors/neoextractors/spikeglx.py", line 50, in __init__
    NeoBaseRecordingExtractor.__init__(
  File "/home/pierre/github/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py", line 210, in __init__
    stream_id = stream_ids[0]
IndexError: list index out of range

Spike sorting failed. You can inspect the runtime trace in sorting/spikeinterface_log.json.

The problem disapears if I am using the full recording (rec instead of rec_sliced)

alejoe91, 2023-07-17:
I'll test today!

yger, 2023-07-17:
Cool. And maybe this is related, but in fact code that used to work with .h5 generated via MEARec is no longer working with the run_sorter() command. 

In fact, as long as I am using the plain recording, everything is fine. But if I am using a preprocessed recording, it crashes

recording, sorting = si.read_mearec('recording_test.h5')
recording_f = si.common_reference(recording)
recording_f = si.highpass_filter(recording_f, dtype='float32', freq_min=150)
recording_f = si.zscore(recording_f, dtype='float32')

res_circus_2 = si.run_sorter('spykingcircus2', recording) works

But 

res_circus_2 = si.run_sorter('spykingcircus2', recording_f) crashes, as if the code was looking for the recoding file in a wrong folder (because it should in circus2, not in circus2/spykingcircus2_output)

FileNotFoundError: [Errno 2] Unable to synchronously open file (unable to open file: name = '/home/pierre/data/sorting/circus2/spykingcircus2_output/recording_test.h5', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0)

Spike sorting failed. You can inspect the runtime trace in spykingcircus2_output/spikeinterface_log.json.

samuelgarcia, 2023-07-17:
Salut Pierre, do you have nei instaled from source ?

alejoe91, 2023-07-17:
can you share the full error trace for the second example? 

Also the full path for the `recording_test.h5`

yger, 2023-07-24:
Closed with #1834

closed 2023-07-24 by yger