SpikeInterface/spikeinterface #1842: Failed to get result with sorter specific extractor.

RikkelBob, 2023-07-18:
I'm having a problem with spikeinterface + tridesclous, but it's hard to pinpoint what is the actual issue.
I've set up a script to do batch processing of a bunch of recordings, which runs fine. However, the output doesn't make any sense (either zero units found, or a bunch of "fake" units, with just noise labeled as spikes).

When sorting has finished, the error below is thrown. Does this mean spikeinterface first tries to retrieve the results one way, fails, and then succeeds another way?

Another error I used to see is that there excess spikes outside of the recording timeframe, which suggests I should use remove_excess_spikes. This solves it, but still worries me. What could be root of this error? Why would tridesclous find spikes outside of my recording?

```
Failed to get result with sorter specific extractor.
Error Message: unsupported operand type(s) for +: 'builtin_function_or_method' and 'builtin_function_or_method'
Getting result from in-container saved NpzSortingExtractor
```

```
import spikeinterface as si
import spikeinterface.preprocessing as spre
import spikeinterface.sorters as ss
import spikeinterface.widgets as sw
import spikeinterface.curation as sic
import utils
import matplotlib.pyplot as plt
from pathlib import Path
import os
import glob
import shutil
from spikeinterface.exporters import export_to_phy
from spikeinterface.exporters import export_report
from spikeinterface.postprocessing import compute_spike_amplitudes, compute_principal_components, compute_template_metrics
from gc import collect

#Some parameters
animals = ["006", "007", "008", "015"]
experiment = "exp2"
sorter_name = "tridesclous"
snippet = True
reorder = True
plot_timeseries = False
plot_probe = False
probe, positions = utils.gen_probe(plot_probe)
print(probe)

#Don't change these
number = 0
fs = 32000
n_chan = 64
file_name_format = f"ksData_probe{number}.dat"
dtype_rec = 'int16'
dtype_save = 'float32'
root_path_nas = "//NAS/myname/AcquisitionPC/"
temp_path = "C:/spikeinterface_temp/"
output_folder = temp_path + sorter_name + "-results"
file_path_pre = temp_path + "preprocessed"
phy_path = temp_path + "phy-output"

for animal in animals:
    print(animal)
    sessions = utils.get_directories(Path(root_path_nas,experiment,animal))
    for session in sessions:
        print(session)
        session_path = Path(root_path_nas,experiment,animal,session)
        dat_files = glob.glob(str(session_path) + '/*.dat')
        i = 1
        for probe_rec in dat_files:
            print(probe_rec)
            probe_dir = "probe" + str(i)
            phy_sorter_dir = Path(Path(root_path_nas, experiment, animal, session, sorter_name))
            if not os.path.isdir(phy_sorter_dir):
                os.mkdir(phy_sorter_dir)

            phy_output_dir = Path(root_path_nas, experiment, animal, session, sorter_name, probe_dir)

            if os.path.isdir(phy_output_dir):
                print(str(phy_output_dir) + " already exists, skipping")
                i += 1
                continue

            print("Moving file to " + temp_path)
            shutil.copyfile(probe_rec,Path(temp_path,"tempdata.dat"))
            recording = si.read_binary((temp_path + "tempdata.dat"), sampling_frequency=fs, num_channels=n_chan, dtype=dtype_rec)

            if snippet:
                start_time_s = 0  # start time in seconds
                end_time_s = 60  # end time in seconds
                start_frame = start_time_s * fs  # start frame
                end_frame = end_time_s * fs  # end frame
                # Create the new recording
                recording = recording.frame_slice(start_frame, end_frame)

            recording.annotate(is_filtered=False)
            recording.set_probe(probe, in_place=True)

            channel_ids = recording.get_channel_ids()
            fs = recording.get_sampling_frequency()
            num_chan = recording.get_num_channels()
            num_segments = recording.get_num_segments()

            print(f'Channel ids: {channel_ids}')
            print(f'Sampling frequency: {fs}')
            print(f'Number of channels: {num_chan}')
            print(f"Number of segments: {num_segments}")

            recording_f = spre.bandpass_filter(recording, freq_min=300, freq_max=6000)
            recording_cmr = spre.common_reference(recording_f, operator="median", reference="global")
            recording_whitened = spre.whiten(recording_cmr,dtype=dtype_save)

            if plot_timeseries:
                w_ts = sw.plot_timeseries(recording, order_channel_by_depth=True,
                                          channel_ids=probe.device_channel_indices)
                w_f = sw.plot_timeseries(recording_f, order_channel_by_depth=True)
                w_cmr = sw.plot_timeseries(recording_cmr, order_channel_by_depth=True)
                w_whitened = sw.plot_timeseries(recording_cmr, order_channel_by_depth=True)
                plt.show()

            recording_saved = recording_whitened.save(folder=file_path_pre, n_jobs=1, format='binary')
            recording_saved.set_probe(probe, in_place=True)

            sorting = ss.run_sorter(sorter_name=sorter_name,
                                    recording=recording_saved,
                                    docker_image=True,
                                    verbose=True) # add num_workers?

            if sorting.get_num_units() == 0:
                print("No clustered units - Skipping export to phy")
                # Delete temp files
                print("Deleting temp files")
                os.mkdir(phy_output_dir)
                del recording, recording_saved, recording_cmr, recording_whitened, recording_f
                del sorting
                collect()
                shutil.rmtree(temp_path)
                os.mkdir(temp_path)
                i += 1
                continue

            #sorting = sorting.save(folder=output_folder)
            #sorting_re = si.load_extractor(output_folder)
            sorting_re = sic.remove_excess_spikes(sorting, recording_saved)

            print("Generating WaveformExtractor")
            we_folder = temp_path + "waveforms"
           # si.curation.remove_excess_spikes(sorting, recording_saved)
            we = si.extract_waveforms(recording_saved, sorting_re, we_folder, sparse=True, dtype='float32')
            print(we)
            # some computations are done before to control all options
            compute_spike_amplitudes(we)
            compute_principal_components(we, n_components=3, mode='by_channel_global')
            compute_template_metrics(we)

            #Remove zero-amplitude units
            unit_ids = sorting_re.get_unit_ids()
            units_to_keep = []
            for unit_id in unit_ids:
                wf = we.get_waveforms(unit_id=unit_id)
                if wf.min() != 0 and wf.max() != 0:
                    units_to_keep.append(unit_id)

            sorting_subset = sorting_re.select_units(unit_ids=units_to_keep)

            we_sub_folder = temp_path + "waveforms_subset"
            we_sub = si.extract_waveforms(recording_saved, sorting_subset, we_sub_folder, sparse=True, dtype='float32')

            # Export
            export_to_phy(we_sub, output_folder=phy_path)

            # Move results back to NAS
            print("Moving phy output to NAS")
            probe_dir = "probe" + str(i)
            shutil.copytree(phy_path, phy_output_dir)

            # Delete temp files
            print("Deleting temp files")
            del recording, recording_saved, recording_cmr, recording_whitened, recording_f
            del we, we_sub
            del sorting, sorting_subset, sorting_re
            del wf
            collect()
            shutil.rmtree(temp_path)
            os.mkdir(temp_path)
            print('Finished ' + probe_rec)
            i += 1

```

RikkelBob, 2023-07-18:
Playing around with it now, some sessions get sorted properly, but the majority is still terrible. The bad ones give me errors about spikes being after the end of the recording

alejoe91, 2023-07-19:
Is that the full error trace?

alejoe91, 2023-07-19:
I don't think Tridesclous like whitened data. Can you try without?

RikkelBob, 2023-07-19:
Please disregard, it turned out I was running an old version of SI, which apparently isn't compatible with the docker image.

closed 2023-07-19 by RikkelBob