SpikeInterface/spikeinterface #1867: does klustakwik work with the listed docker file?

SusanL82, 2023-07-21:
Hello, 

After spending about a week trying to install klusta/klustakwik with spikeinterface, I was able to install all the required packages but not use them in spyder (spikeinterface insists that the sorter is not installed). I know klusta is a legacy package, so I don't expect help regarding this. 

My hope is to use spikeinterface to read NLX recordings, cluster using klustakwik, export to phy for manual curation. 

So far, I've found that klusta is the only sorter that will perform well on CA3 tetrode recordings (kilosort fails, and I've been struggling with MS4). Pretty much every smarter sorter seems to have issues with the complex spikes from this area and with the remapping characteristics of my cells (e.g. cells can be quiet for 10min when the rat is not in an environment those cells code for). 

I am now hoping that I can use the klusta docker image instead. Could someone please confirm that this works before I spend another week trying to figure out how to use docker?

alejoe91, 2023-07-21:
Hi @SusanL82 

It should work if you use SI<0.96.0. After that SI only supports Python 3.8, which is incompatible with klusta. 

Have you tried `Waveclus`? It should also work ok with tetrodes!

SusanL82, 2023-07-24:
Hi!
I managed to get things to work with SI 0.95.1 and Python 3.7. I think Spyder and its widgets gave me all my issues actually (not SI). 
I'll keep WaveClus in mind for future options, though. Thanks!

closed 2023-07-24 by SusanL82