SpikeInterface/spikeinterface #1869: KeyError: 'pca_1023' in compute_quality_metrics

ChenLuisme, 2023-07-21:
When I try to compute the quality metrics (I try to run all possible metrics to have a basic idea how my data look like), I met a error called KeyError: 'pca_1023'

The code
`new_m = si.compute_quality_metrics(we,n_jobs=10,progress_bar=True)`

The error
```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Cell In[32], line 1
----> 1 new_m = si.compute_quality_metrics(we,n_jobs=10,progress_bar=True)

File [c:\Users\LindenLab\anaconda3\envs\latte\lib\site-packages\spikeinterface\qualitymetrics\quality_metric_calculator.py:231](file:///C:/Users/LindenLab/anaconda3/envs/latte/lib/site-packages/spikeinterface/qualitymetrics/quality_metric_calculator.py:231), in compute_quality_metrics(waveform_extractor, load_if_exists, metric_names, qm_params, peak_sign, seed, sparsity, skip_pc_metrics, verbose, **job_kwargs)
    222     qmc = QualityMetricCalculator(waveform_extractor)
    223     qmc.set_params(
    224         metric_names=metric_names,
    225         qm_params=qm_params,
   (...)
    229         skip_pc_metrics=skip_pc_metrics,
    230     )
--> 231     qmc.run(verbose=verbose, **job_kwargs)
    233 metrics = qmc.get_data()
    235 return metrics

File [c:\Users\LindenLab\anaconda3\envs\latte\lib\site-packages\spikeinterface\core\waveform_extractor.py:1841](file:///C:/Users/LindenLab/anaconda3/envs/latte/lib/site-packages/spikeinterface/core/waveform_extractor.py:1841), in BaseWaveformExtractorExtension.run(self, **kwargs)
   1840 def run(self, **kwargs):
-> 1841     self._run(**kwargs)
   1842     self._save(**kwargs)

File [c:\Users\LindenLab\anaconda3\envs\latte\lib\site-packages\spikeinterface\qualitymetrics\quality_metric_calculator.py:136](file:///C:/Users/LindenLab/anaconda3/envs/latte/lib/site-packages/spikeinterface/qualitymetrics/quality_metric_calculator.py:136), in QualityMetricCalculator._run(self, verbose, **job_kwargs)
    134     raise ValueError("waveform_principal_component must be provied")
    135 pc_extension = self.waveform_extractor.load_extension("principal_components")
--> 136 pc_metrics = calculate_pc_metrics(
    137     pc_extension,
    138     unit_ids=non_empty_unit_ids,
    139     metric_names=pc_metric_names,
    140     sparsity=sparsity,
    141     progress_bar=progress_bar,
    142     n_jobs=n_jobs,
    143     qm_params=qm_params,
    144     seed=seed,
    145 )
    146 for col, values in pc_metrics.items():
    147     metrics.loc[non_empty_unit_ids, col] = pd.Series(values)

File [c:\Users\LindenLab\anaconda3\envs\latte\lib\site-packages\spikeinterface\qualitymetrics\pca_metrics.py:138](file:///C:/Users/LindenLab/anaconda3/envs/latte/lib/site-packages/spikeinterface/qualitymetrics/pca_metrics.py:138), in calculate_pc_metrics(pca, metric_names, sparsity, qm_params, unit_ids, seed, n_jobs, progress_bar)
    135 if run_in_parallel:
    136     parallel_functions = []
--> 138 all_labels, all_pcs = pca.get_all_projections()
    139 for unit_ind, unit_id in units_loop:
    140     if we.is_sparse():

File [c:\Users\LindenLab\anaconda3\envs\latte\lib\site-packages\spikeinterface\postprocessing\principal_component.py:137](file:///C:/Users/LindenLab/anaconda3/envs/latte/lib/site-packages/spikeinterface/postprocessing/principal_component.py:137), in WaveformPrincipalComponent.get_all_projections(self, channel_ids, unit_ids, outputs)
    135 all_projections = []
    136 for unit_index, unit_id in enumerate(unit_ids):
--> 137     proj = self.get_projections(unit_id)
    138     if channel_ids is not None:
    139         chan_inds = self.waveform_extractor.channel_ids_to_indices(channel_ids)

File [c:\Users\LindenLab\anaconda3\envs\latte\lib\site-packages\spikeinterface\postprocessing\principal_component.py:89](file:///C:/Users/LindenLab/anaconda3/envs/latte/lib/site-packages/spikeinterface/postprocessing/principal_component.py:89), in WaveformPrincipalComponent.get_projections(self, unit_id)
     75 def get_projections(self, unit_id):
     76     """
     77     Returns the computed projections for the sampled waveforms of a unit id.
     78 
   (...)
     87         The PCA projections (num_waveforms, num_components, num_channels)
     88     """
---> 89     return self._extension_data[f"pca_{unit_id}"]

KeyError: 'pca_1023'

```

alejoe91, 2023-07-21:
Hi,

How do you compute PCA projections?

Maybe there's something wrong there!

ChenLuisme, 2023-07-21:
I thought that this code would compute both PCA and non-PCA quality metrics. Sorry for the naive question: do I need to pre-compute PCA projections first? I am not sure how this work.

alejoe91, 2023-07-21:
It should compute PCA based quality metrics if you have computed them first. You can check out this tutorial on extensions: https://spikeinterface.readthedocs.io/en/latest/modules/postprocessing.html

h-mayorquin, 2023-07-26:
@ChenLuisme did this solve your error?

@alejoe91 would it be possible to have a more informative error that tell users they have to pre-compute this?

alejoe91, 2023-07-26:
@h-mayorquin PCA metrics are computed only if PCA has been pre-computed. That's why this error looks strange. @ChenLuisme did you manipulate the waveform folder?

alejoe91, 2023-11-06:
@ChenLuisme any update on this?

ChenLuisme, 2023-11-06:
Hi,

It got solved! I should calculate the PCA first.
Many thanks.

Best wishes,
Chen

Chen Lu
she/her
University College London
Ear Institute
332 Gray's Inn Road
London, WC1X 8EE
Tel: +44 (0)75 2993 2905
Email: ***@***.***

⚠ Caution: External sender

closed 2023-11-06 by alejoe91