SpikeInterface/spikeinterface #1870: Numpy 1.24 breaks a few sorters

jonahpearl, 2023-07-21:
Hi there -- I'm installing spikeinterface today and noticed a small issue: the removal of the numpy alias `np.bool` breaks `herdingspikes` in addition to `trisdeclous` (as noted in the install cheat sheet ymls). (This is not the only popular repo where this alias has caused problems...) Maybe consider pinning `numpy=1.22` so that new spikeinterface installs don't hit this issue, while spike sorter maintainers have a chance to fix this bug, unless the last two versions of numpy really provide some major / required improvements?

h-mayorquin, 2023-07-21:
I guess that @samuelgarcia can fix `tridesclous` but the last release of herdingspikes was three years ago.

alejoe91, 2023-07-21:
Thanks for reporting @jonahpearl 

I think `tridesclous.1.6.8` (release 2 days ago) is updated, since it passes tests on our side too. 

For herdingspikes, as @h-mayorquin says, they should fix it on their end. We don't want to pin numpy because a specific sorter doesn't support a certain version. The user can either handle the upgrades/downgrades or run with docker/singularity (recommended!)

h-mayorquin, 2023-07-21:
Maybe somebody can convince Mathias to make some updates. I remember other deprecations and problems over there. We needed to import something prebuilding or something like that. But PI life is a busy one (not being sarcastic here!).

h-mayorquin, 2023-07-26:
@jonahpearl can we close this issue? Is there anything on the documentation you think could be improved to indicate the state of affaris? 

Right now it someone try to use `herdingspikes` what happens? Is the error informative of what is the problem and what needs to be done?

jonahpearl, 2023-07-26:
Good question, thanks. I encountered the issue when using the [`check_your_install.py`](https://github.com/SpikeInterface/spikeinterface/blob/main/installation_tips/check_your_install.py) script. When it failed, there was no error message other than "failed", so I went and ran it manually to see the error message. From there it was pretty clear that I had to downgrade `numpy` (though `herdingspikes` at this point requires a pretty specific `numpy` version due to [this error](https://github.com/mhhennig/HS2/issues/58#issuecomment-1645986966) which requires *up*grading it past a certain point, so that was a bit frustrating). 

I guess you might consider adding error logging to the check install script, or a note about needing to fix `numpy` if you're using `herdingspikes`, but I can also see the argument that if someone isn't savvy enough to solve those problems, they should perhaps just use docker. Hopefully the sorter just gets fixed -- it should be (theoretically) as easy as using find and replace from `np.bool` to `bool`. Feel free to close.

h-mayorquin, 2023-07-26:
Thanks, this is very useful.  I was not even aware that we have that script. Maybe we should test that so we don't give wrong installation instructions. It seems like a big diservice to the less experienced users that are more likely to use that script.

Anyway, I think that we should 1) modify the `check_your_install` to account for this new reality (either add a version check or remove the herding spikes from there) and 2) if possible add a test for the whole `check_your_install` or rethink how we handle our installation instructions so they don't drift away from the state of our repo as they have done here.

zm711, 2023-11-03:
We have updated check_your_install.py, added exception messaging in case any step fails, and added a cron job test to make sure check install works. But if anything else comes up just open a new issue.

closed 2023-11-03 by zm711