SpikeInterface/spikeinterface #1871: Error running run_sorter in py script from command prompt

oghenand, 2023-07-21:
Hi! 

I'm trying to run Ironclust using Spike Interface on a .py file that I want to directly run from the command prompt. However, when I try to do so, I get an odd error likely due to saving of the data to the output folder. When I run the code from jupyter notebook, it runs completely fine, but the error seems to stem from using a .py file. I've attached the output of my error message below. If anyone has encountered a similar error before and would know a fix, that'd be great, thanks!

`Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\ProgramData\miniconda3\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\ProgramData\miniconda3\lib\multiprocessing\spawn.py", line 125, in _main
    prepare(preparation_data)
  File "C:\ProgramData\miniconda3\lib\multiprocessing\spawn.py", line 236, in prepare
    _fixup_main_from_path(data['init_main_from_path'])
  File "C:\ProgramData\miniconda3\lib\multiprocessing\spawn.py", line 287, in _fixup_main_from_path
    main_content = runpy.run_path(main_path,
  File "C:\ProgramData\miniconda3\lib\runpy.py", line 289, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\ProgramData\miniconda3\lib\runpy.py", line 96, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "C:\ProgramData\miniconda3\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\Allison\source\repos\Spike_Sorting_Automation_1\Spike_Sorting_Automation_1\Spike_Sorting_Automation_1.py", line 149, in <module>
    IC_si = ss.run_sorter('ironclust', recording = recording_prb, output_folder = r'C:\Users\Allison\Desktop\auto_pilot_721', verbose=False)
  File "C:\Users\Allison\AppData\Roaming\Python\Python310\site-packages\spikeinterface\sorters\runsorter.py", line 143, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "C:\Users\Allison\AppData\Roaming\Python\Python310\site-packages\spikeinterface\sorters\runsorter.py", line 158, in run_sorter_local
    output_folder = SorterClass.initialize_folder(
  File "C:\Users\Allison\AppData\Roaming\Python\Python310\site-packages\spikeinterface\sorters\basesorter.py", line 120, in initialize_folder
    shutil.rmtree(str(output_folder))
  File "C:\ProgramData\miniconda3\lib\shutil.py", line 750, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\ProgramData\miniconda3\lib\shutil.py", line 615, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\ProgramData\miniconda3\lib\shutil.py", line 615, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\ProgramData\miniconda3\lib\shutil.py", line 620, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\ProgramData\miniconda3\lib\shutil.py", line 618, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\Allison\\Desktop\\auto_pilot_721\\sorter_output\\ironclust_dataset\\raw.mda'
Running ironclust...
Running ironclust...
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\ProgramData\miniconda3\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\ProgramData\miniconda3\lib\multiprocessing\spawn.py", line 125, in _main
    prepare(preparation_data)
  File "C:\ProgramData\miniconda3\lib\multiprocessing\spawn.py", line 236, in prepare
    _fixup_main_from_path(data['init_main_from_path'])
  File "C:\ProgramData\miniconda3\lib\multiprocessing\spawn.py", line 287, in _fixup_main_from_path
    main_content = runpy.run_path(main_path,
  File "C:\ProgramData\miniconda3\lib\runpy.py", line 289, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\ProgramData\miniconda3\lib\runpy.py", line 96, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "C:\ProgramData\miniconda3\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\Allison\source\repos\Spike_Sorting_Automation_1\Spike_Sorting_Automation_1\Spike_Sorting_Automation_1.py", line 149, in <module>
    IC_si = ss.run_sorter('ironclust', recording = recording_prb, output_folder = r'C:\Users\Allison\Desktop\auto_pilot_721', verbose=False)
  File "C:\Users\Allison\AppData\Roaming\Python\Python310\site-packages\spikeinterface\sorters\runsorter.py", line 143, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "C:\Users\Allison\AppData\Roaming\Python\Python310\site-packages\spikeinterface\sorters\runsorter.py", line 158, in run_sorter_local
    output_folder = SorterClass.initialize_folder(
  File "C:\Users\Allison\AppData\Roaming\Python\Python310\site-packages\spikeinterface\sorters\basesorter.py", line 120, in initialize_folder
    shutil.rmtree(str(output_folder))
  File "C:\ProgramData\miniconda3\lib\shutil.py", line 750, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\ProgramData\miniconda3\lib\shutil.py", line 615, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\ProgramData\miniconda3\lib\shutil.py", line 615, in _rmtree_unsafe
[... 51 lines cut ...]
  File "C:\Users\Allison\AppData\Roaming\Python\Python310\site-packages\spikeinterface\sorters\runsorter.py", line 143, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "C:\Users\Allison\AppData\Roaming\Python\Python310\site-packages\spikeinterface\sorters\runsorter.py", line 162, in run_sorter_local
    SorterClass.setup_recording(recording, output_folder, verbose=verbose)
  File "C:\Users\Allison\AppData\Roaming\Python\Python310\site-packages\spikeinterface\sorters\basesorter.py", line 196, in setup_recording
    cls._setup_recording(recording, sorter_output_folder, sorter_params, verbose)
  File "C:\Users\Allison\AppData\Roaming\Python\Python310\site-packages\spikeinterface\sorters\external\ironclust.py", line 169, in _setup_recording
    MdaRecordingExtractor.write_recording(recording=recording, save_path=str(dataset_dir), verbose=False,
  File "C:\Users\Allison\AppData\Roaming\Python\Python310\site-packages\spikeinterface\extractors\mdaextractors.py", line 118, in write_recording
    write_binary_recording(recording, file_paths=save_file_path, dtype=dtype,
  File "C:\Users\Allison\AppData\Roaming\Python\Python310\site-packages\spikeinterface\core\core_tools.py", line 280, in write_binary_recording
    executor.run()
  File "C:\Users\Allison\AppData\Roaming\Python\Python310\site-packages\spikeinterface\core\job_tools.py", line 364, in run
    for res in results:
  File "C:\ProgramData\miniconda3\lib\concurrent\futures\process.py", line 570, in _chain_from_iterable_of_lists
    for element in iterable:
  File "C:\ProgramData\miniconda3\lib\concurrent\futures\_base.py", line 621, in result_iterator
    yield _result_or_cancel(fs.pop())
  File "C:\ProgramData\miniconda3\lib\concurrent\futures\_base.py", line 319, in _result_or_cancel
    return fut.result(timeout)
  File "C:\ProgramData\miniconda3\lib\concurrent\futures\_base.py", line 458, in result
    return self.__get_result()
  File "C:\ProgramData\miniconda3\lib\concurrent\futures\_base.py", line 403, in __get_result
    raise self._exception
concurrent.futures.process.BrokenProcessPool: A process in the process pool was terminated abruptly while the future was running or pending.`

oghenand, 2023-07-25:
Hi! 

How do you transform the binary extractor? I'm using an intan rhd recording for the analysis, btw. 

I read the issue you linked, and I can try updating the package if that will help? Let me know if there's anything else you suggest I can do! 

I found a work-around running the code in an ipynb file, but getting it to run in a py file would be ideal!

Thanks!

h-mayorquin, 2023-07-26:
From your recorder you just do 
```python
cached_recording = your_recording.save() 
```

This will create a copy of your recorder saved in your computer. So there is some extra space that will be used. But now that I think twice about it, I am less sure this will solve your problem since I don't think that intan should have any problems. 

Can you create to run your script with the following recording just to see if we can reproduce your error with something that you can run:

```python
from spikeinterface.core.generate import generate_lazy_recording

recording = generate_lazy_recording(full_traces_size_GiB=1.0) 
recording
```

Please share your script and instead of using your recorder use the one that I just provided. If the error is reproduced then we can debug easier on our side.

KLehongre, 2023-08-24:
Hello, 

I have the same issue, or at least I think it is the same, when running spykingcircus2 on the dev version ('0.99.0.dev0'), with anaconda on Windows.
I do not have the issue with the pip installation ('0.98.2').
Did you find a solution? I've tried to save the recording but it did not make any difference.
Here is the error message:
SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\si_dev\Lib\site-packages\spikeinterface-0.99.0.dev0-py3.11.egg\spikeinterface\sorters\basesorter.py", line 234, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\si_dev\Lib\site-packages\spikeinterface-0.99.0.dev0-py3.11.egg\spikeinterface\sorters\internal\spyking_circus2.py", line 112, in _run_from_folder
    labels, peak_labels = find_cluster_from_peaks(
                          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\si_dev\Lib\site-packages\spikeinterface-0.99.0.dev0-py3.11.egg\spikeinterface\sortingcomponents\clustering\main.py", line 41, in find_cluster_from_peaks
    labels, peak_labels = method_class.main_function(recording, peaks, params)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\si_dev\Lib\site-packages\spikeinterface-0.99.0.dev0-py3.11.egg\spikeinterface\sortingcomponents\clustering\random_projections.py", line 242, in main_function
    shutil.rmtree(tmp_folder / "sorting")
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\si_dev\Lib\shutil.py", line 759, in rmtree
    return _rmtree_unsafe(path, onerror)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\si_dev\Lib\shutil.py", line 622, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\si_dev\Lib\shutil.py", line 620, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 32] Le processus ne peut pas accéder au fichier car ce fichier est utilisé par un autre processus: '\\\\l2export\\iss02.epimicro\\patients\\shared\\Virginie\\SLI\\SLI_analysis_spikes\\data\\SI\\02420\\p1\\mOcp\\SI_pos_th6_rec\\0\\spykingcircus2\\sorter_output\\clustering\\sorting\\spikes.npy'

Thanks!

yger, 2023-08-24:
Not sure this is the same bug, but maybe. Yours seems to be very much related to internal machinery of SC2, I'll try to have a look quickly

LeMuellerGuy, 2023-09-26:
I have encountered the same problem running spyking circus 2 on windows (running 0.99.0.dev0). I have narrowed it down to this section of code in the `random_projections.py` file (lines 196 - 213):
```
elif cleaning_method == "matching":
    # create a tmp folder
    if params["tmp_folder"] is None:
        name = "".join(random.choices(string.ascii_uppercase + string.digits, k=8))
        tmp_folder = get_global_tmp_folder() / name
    else:
        tmp_folder = Path(params["tmp_folder"])

    if params["shared_memory"]:
        waveform_folder = None
        mode = "memory"
    else:
        waveform_folder = tmp_folder / "waveforms"
        mode = "folder"

    sorting_folder = tmp_folder / "sorting"
    sorting = NumpySorting.from_times_labels(spikes["sample_index"], spikes["unit_index"], fs)
    sorting = sorting.save(folder=sorting_folder)
    we = extract_waveforms(
                recording,
                sorting,
                waveform_folder,
                ms_before=params["ms_before"],
                ms_after=params["ms_after"],
                **params["job_kwargs"],
                return_scaled=False,
                mode=mode,
            )
```
as well as 
```
labels, peak_labels = remove_duplicates_via_matching(
                we, noise_levels, peak_labels, job_kwargs=cleaning_matching_params, **cleaning_params
            )
```
In my specific scenario I have a tmp_folder location already set and am not using shared memory.
Thus it saves a spikes.npy file in base_folder/sorter_output/clustering/sorting
Later on in the code there is a section where it tries to remove said folder.
```
if params["tmp_folder"] is None:
    shutil.rmtree(tmp_folder)
else:
    
    shutil.rmtree(tmp_folder / "waveforms")
    # this line causes the crash because it attempts to delete the sorting folder. 
    # However, the spikes.npy file is still referenced by the `sorting` object created earlier.
    shutil.rmtree(tmp_folder / "sorting") 
```
It is apparent that somewhere it opens the spikes.py file for streaming or it isn't close completely. 
I was able to delete the directories before calling `remove_duplicates_via_matching()` by manually calling `del` for the sorting and we object. However, I was unable to delete them after the call. I thus believe that somewhere in `remove_duplicates_via_matching()` another reference to the file is created. I was unable to find it though. I have tried using both the garbage collector to cut any lose ties and manually deleting all local variables (even the ones that are actually returned by the function) before trying to remove the directories but this did not have the desired effect and still ended with a WinError 32 because of spikes.npy.
I also don't see any way to really combat this without a deep understanding of the issue since the spikes.npy file is created regardless of any settings that the user can make.

samuelgarcia, 2023-09-27:
Thanks all for reporting this.
Definitivelly we need to handle very carffully the memmaped files after use, especially when we want to delete then.

h-mayorquin, 2023-09-28:
I also have the suspicion that some of these problems come from decision of handling the creation and deletion of temporary files.

yger, 2023-09-29:
Thanks for the feedback, and indeed, there is something fishy with the folders here, I can not understand why it can not be removed properly. One way to deal with that is to add ignore_errors=True to rmtree(), but it only solves partially the problem

zm711, 2023-12-19:
I'm wondering if anyone has tested re-running their analyses. Spikeinterface (0.99.1 & 0.100.0dev--especially 0.100.0dev) have had some additions/fixes to make working on Windows better. It would be great to know if anyone has tried using the package since these updates so we can see if real world cases have been improved.

LeMuellerGuy, 2024-01-05:
I reran spykingcircus2 on the 0.100.0.dev version but the issue is still present

zm711, 2024-01-05:
Can you copy the exact error that is appearing below?

LeMuellerGuy, 2024-01-05:
On line 126 in basesorter.py:
Translation of error message: The process can't access the file because it is being used by another process

```
Exception has occurred: PermissionError
[WinError 32] Der Prozess kann nicht auf die Datei zugreifen, da sie von einem anderen Prozess verwendet wird: 'C:\\Users\\lennart\\sciebo\\Masterarbeit\\CodeProjects\\spikeinterface\\spykingcircus2_output\\sorter_output\\sorting\\spikes.npy'
  File "C:\Users\lennart\spikeinterface\src\spikeinterface\sorters\basesorter.py", line 126, in initialize_folder
    shutil.rmtree(str(output_folder))
  File "C:\Users\lennart\spikeinterface\src\spikeinterface\sorters\runsorter.py", line 171, in run_sorter_local
    output_folder = SorterClass.initialize_folder(recording, output_folder, verbose, remove_existing_folder)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\lennart\spikeinterface\src\spikeinterface\sorters\runsorter.py", line 148, in run_sorter
    return run_sorter_local(**common_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\lennart\sciebo\Masterarbeit\CodeProjects\spikeinterface\AxonTracking\concatenateRecordings.py", line 52, in <module>
PermissionError: [WinError 32] Der Prozess kann nicht auf die Datei zugreifen, da sie von einem anderen Prozess verwendet wird: 
'C:\\Users\\lennart\\sciebo\\Masterarbeit\\CodeProjects\\spikeinterface\\spykingcircus2_output\\sorter_output\\sorting\\spikes.npy'
```

zm711, 2024-01-05:
Thanks @LeMuellerGuy . Yeah I think the base issue affecting all sorters has been fixed. Would you be willing to test any other sorter so we can confirm this is isolated to spykingcircus2 now?

yger, 2024-01-05:
I'll have a look and discuss that with @samuelgarcia next week. Since I am not using windows, I never encountered these issues....

zm711, 2024-01-05:
@yger, I'm on windows currently so if you need me to run any tests for SC2 feel free to shoot me a slack (but I know Sam is doing dual-boot currently so he can also help with all the tests). But these permission errors come up all the time because Windows tries to delete files right away with shutil rather than put them in a queue to be deleted later. Sam fixed a bunch of these using some `weakref` and tweaking some memory mapping of files (so he can explain better), but I think SC2 using shutil.rm internally right?

LeMuellerGuy, 2024-01-05:
I will have a deeper look at things tomorrow since it is quite late here already. I'll test some other sorters too.

yger, 2024-01-05:
Yes, there are some internal calls to shutil.rmtree in SC2.... I could try to get rid of them maybe, i'll also have a look tomorrow. If you want to have a go, this is now easy with the GT recordings generated on the fly. Let me know if the following lines are crashing on windows

rec, sorting = si.generate_ground_truth_recording(num_channels=32)
sorting = si.run_sorter('spykingcircus2', rec, remove_existing_folder=True)

zm711, 2024-01-05:
So on the first try it runs fine without a problem. 

```python
sorting = si.run_sorter('spykingcircus2', rec, remove_existing_folder=True, output_folder='test-yger', verbose=True)
detect peaks using locally_exclusive with n_jobs = 8 and chunk_size = 25000
detect peaks using locally_exclusive:   0%|          | 0/10 [00:00<?, ?it/s]
We found 1239 peaks in total
We kept 1239 peaks for clustering
extracting features with n_jobs = 8 and chunk_size = 25000
extracting features:   0%|          | 0/10 [00:00<?, ?it/s]
We found 5 raw clusters, starting to clean with matching...
extract waveforms shared_memory multi buffer with n_jobs = 8 and chunk_size = 25000
extract waveforms shared_memory multi buffer:   0%|          | 0/10 [00:00<?, ?it/s]
extract waveforms shared_memory multi buffer with n_jobs = 8 and chunk_size = 25000
extract waveforms shared_memory multi buffer:   0%|          | 0/10 [00:00<?, ?it/s]
We kept 5 non-duplicated clusters...
extract waveforms shared_memory multi buffer with n_jobs = 8 and chunk_size = 25000
extract waveforms shared_memory multi buffer:   0%|          | 0/10 [00:00<?, ?it/s]
extract waveforms shared_memory multi buffer with n_jobs = 8 and chunk_size = 25000
extract waveforms shared_memory multi buffer:   0%|          | 0/10 [00:00<?, ?it/s]
find spikes (circus-omp-svd) with n_jobs = 8 and chunk_size = 2500
find spikes (circus-omp-svd):   0%|          | 0/100 [00:00<?, ?it/s]
We found 1063 spikes
spykingcircus2 run time 107.89s
```

If I then rerun I get the same error as above: 

```python

  Cell In[160], line 1
    sorting = si.run_sorter('spykingcircus2', rec, remove_existing_folder=True, output_folder='test-yger', verbose=True)

  File ~\Documents\GitHub\spikeinterface\src\spikeinterface\sorters\runsorter.py:148 in run_sorter
    return run_sorter_local(**common_kwargs)

  File ~\Documents\GitHub\spikeinterface\src\spikeinterface\sorters\runsorter.py:171 in run_sorter_local
    output_folder = SorterClass.initialize_folder(recording, output_folder, verbose, remove_existing_folder)

  File ~\Documents\GitHub\spikeinterface\src\spikeinterface\sorters\basesorter.py:126 in initialize_folder
    shutil.rmtree(str(output_folder))

  File ~\anaconda3\envs\spykes\Lib\shutil.py:759 in rmtree
    return _rmtree_unsafe(path, onerror)

  File ~\anaconda3\envs\spykes\Lib\shutil.py:617 in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)

  File ~\anaconda3\envs\spykes\Lib\shutil.py:617 in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)

  File ~\anaconda3\envs\spykes\Lib\shutil.py:622 in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())

  File ~\anaconda3\envs\spykes\Lib\shutil.py:620 in _rmtree_unsafe
    os.unlink(fullname)

PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\ZM\\test-yger\\sorter_output\\sorting\\spikes.npy'
```

LeMuellerGuy, 2024-01-06:
I reran the same code from yesterday evening again and it (thankfully) produced the same result as @zm711. It worked on the first time and crashed on folder initialization the second time.
I also tried Tridesclous2 plugged into the same code which also produced the same error (trace below). It also crashed on the second time running it when trying to overwrite the existing folder. I assume that external sorters will also suffer from the same problem.

```
Exception has occurred: PermissionError
[WinError 32] Der Prozess kann nicht auf die Datei zugreifen, da sie von einem anderen Prozess verwendet wird: 'C:\\Users\\lennart\\sciebo\\Masterarbeit\\CodeProjects\\spikeinterface\\tridesclous2_output\\sorter_output\\sorting\\spikes.npy'
  File "C:\Users\lennart\spikeinterface\src\spikeinterface\sorters\basesorter.py", line 126, in initialize_folder
    shutil.rmtree(str(output_folder))
  File "C:\Users\lennart\spikeinterface\src\spikeinterface\sorters\runsorter.py", line 171, in run_sorter_local
    output_folder = SorterClass.initialize_folder(recording, output_folder, verbose, remove_existing_folder)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\lennart\spikeinterface\src\spikeinterface\sorters\runsorter.py", line 148, in run_sorter
    return run_sorter_local(**common_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\lennart\sciebo\Masterarbeit\CodeProjects\spikeinterface\AxonTracking\concatenateRecordings.py", line 55, in <module>
    multisorting = run_sorter('tridesclous2', concated, remove_existing_folder=True, verbose = True, **params)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [WinError 32] Der Prozess kann nicht auf die Datei zugreifen, da sie von einem anderen Prozess verwendet wird: 'C:\\Users\\lennart\\sciebo\\Masterarbeit\\CodeProjects\\spikeinterface\\tridesclous2_output\\sorter_output\\sorting\\spikes.npy'
```

samuelgarcia, 2024-01-09:
Hi all.
Thanks a lot for this tests.
I guess that the spikes vector from the first run is not copied in memory and stiçll is a memmap which is a bad idea.
I will make some test in this PR #2267.
I have now a windows fully wokring!!

GabrielGaugain, 2024-02-16:
I guess there is still the issue or a related one, trying to run built-in spike sorters give an error related to shared memory:
`  nbytes = int(np.prod(shape) * dtype.itemsize)
Error running tridesclous2
Traceback (most recent call last):
  File "D:\Users\ggaugain\Documents\postdoc\neuron_recordings\test_spikeinterface.py", line 104, in <module>
    sorting = si.run_sorter(sorter_name=sname, recording=recording_saved, output_folder=base_folder /sname, remove_existing_folder=True,verbose=True, **sorter_params)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\sorters\runsorter.py", line 174, in run_sorter
    return run_sorter_local(**common_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\sorters\runsorter.py", line 224, in run_sorter_local
    SorterClass.run_from_folder(output_folder, raise_error, verbose)
  File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\sorters\basesorter.py", line 293, in run_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\sorters\basesorter.py", line 258, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\sorters\internal\tridesclous2.py", line 123, in _run_from_folder
    recording = cache_preprocessing(
                ^^^^^^^^^^^^^^^^^^^^
  File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\sortingcomponents\tools.py", line 92, in cache_preprocessing
    recording = recording.save_to_memory(format="memory", shared=True, **job_kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\core\base.py", line 853, in save_to_memory
    cached = self._save(format="memory", sharedmem=sharedmem, **save_kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\core\baserecording.py", line 490, in _save
    cached = SharedMemoryRecording.from_recording(self, **job_kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\core\numpyextractors.py", line 215, in from_recording
    traces_list, shms = write_memory_recording(source_recording, buffer_type="sharedmem", **job_kwargs)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\core\recording_tools.py", line 322, in write_memory_recording
    arr, shm = make_shared_array(shape, dtype)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\core\core_tools.py", line 167, in make_shared_array
    shm = SharedMemory(name=None, create=True, size=nbytes)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\multiprocessing\shared_memory.py", line 77, in __init__
    raise ValueError("'size' must be a positive integer")

ValueError: 'size' must be a positive integer`

zm711, 2024-02-16:
@GabrielGaugain,

which version of spikeinterface are you using right now? @samuelgarcia and @yger have both done work on tdc2 and sc2 since this issue was first open. So if you haven't updated to the most recent version
```
pip install -U spikeinterface
```

I would start with that and see if that fixes it. If you've already done that then let us know so they can see what is happening.

zm711, 2024-03-22:
Howdy Gabriel!

So TDC2 and SC2 have a caching error that fails on Windows. I have opened the discussion here #2164. Could you post your specific error there. @samuelgarcia and @yger will need to see the specific errors so they can work on fixing it for Windows users. I think we are close.

For Mountainsort5 there is an issue with the cleanup of files which is a known problem for Windows. We are currently deciding how we want to fix in this thread #2607. I can give you a work-around if you're willing to install spikeinterface from source and are using at least python 3.10.

yger, 2024-03-22:
What if you add the option cache_preprocessing={'mode' : None} when you are launching SC2? Does it work? Because  I've encountered similar issues on windows machine with the automated cache.

GabrielGaugain, 2024-03-22:
Same issue, It seems to go deeper when only one processor is used as setting n_jobs to 1 gave another error:
`File "d:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\scipy\signal\_savitzky_golay.py", line 101, in savgol_coeffs
    raise ValueError("polyorder must be less than window_length.")
ValueError: polyorder must be less than window_length.`

Same for TDC2 with the following error:
 `File "d:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\sklearn\utils\validation.py", line 969, in check_array
    raise ValueError(
ValueError: Found array with 0 sample(s) (shape=(0, 20)) while a minimum of 1 is required by `TruncatedSVD.`

Might be specific to my data you think ?

LeMuellerGuy, 2024-03-22:
@GabrielGaugain as far as I know the polyorder error you got running in what I assume is Spyking Circus 2 is related to the savgol filter that (was?) used in it. The window length parameter has to be set according to your data's sampling rate as specified in https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.savgol_filter.html, specifically that the window length must be less or equal to the number of data points and that the polynomial order must be less than the window length. 
Because the parameter of spyking circus sets the window length in milliseconds, it is tied to your sampling rate.
As far as I know the whole savgol part has been removed in the latest version (correct me if I'm wrong here) though.

h-mayorquin, 2024-03-22:
> I guess there is still the issue or a related one, trying to run built-in spike sorters give an error related to shared memory: ` nbytes = int(np.prod(shape) * dtype.itemsize) Error running tridesclous2 Traceback (most recent call last): File "D:\Users\ggaugain\Documents\postdoc\neuron_recordings\test_spikeinterface.py", line 104, in sorting = si.run_sorter(sorter_name=sname, recording=recording_saved, output_folder=base_folder /sname, remove_existing_folder=True,verbose=True, **sorter_params) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\sorters\runsorter.py", line 174, in run_sorter return run_sorter_local(**common_kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\sorters\runsorter.py", line 224, in run_sorter_local SorterClass.run_from_folder(output_folder, raise_error, verbose) File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\sorters\basesorter.py", line 293, in run_from_folder raise SpikeSortingError( spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting error trace: Traceback (most recent call last): File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\sorters\basesorter.py", line 258, in run_from_folder SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose) File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\sorters\internal\tridesclous2.py", line 123, in _run_from_folder recording = cache_preprocessing( ^^^^^^^^^^^^^^^^^^^^ File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\sortingcomponents\tools.py", line 92, in cache_preprocessing recording = recording.save_to_memory(format="memory", shared=True, **job_kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> 
> File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\core\base.py", line 853, in save_to_memory cached = self._save(format="memory", sharedmem=sharedmem, **save_kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> 
> File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\core\baserecording.py", line 490, in _save cached = SharedMemoryRecording.from_recording(self, **job_kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> 
> File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\core\numpyextractors.py", line 215, in from_recording traces_list, shms = write_memory_recording(source_recording, buffer_type="sharedmem", **job_kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> 
> File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\core\recording_tools.py", line 322, in write_memory_recording arr, shm = make_shared_array(shape, dtype) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> 
> File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\site-packages\spikeinterface\core\core_tools.py", line 167, in make_shared_array shm = SharedMemory(name=None, create=True, size=nbytes) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> 
> File "D:\Users\ggaugain\Anaconda3\envs\si_env\Lib\multiprocessing\shared_memory.py", line 77, in **init** raise ValueError("'size' must be a positive integer")
> 
> ValueError: 'size' must be a positive integer`

This error I actually know how to solve. It is caused by that `np.prod`.

I will make a patch.

h-mayorquin, 2024-03-22:
@GabrielGaugain I think this should solve your first error on top:

https://github.com/SpikeInterface/spikeinterface/issues/1871#issuecomment-1947957281

wesleypclawson, 2024-03-22:
Jumping in with a similar poly error as mentioned above:

`ValueError: polyorder must be less than window_length. during feature extraction`

This is a relatively fresh install of SpikeInterface (yesterday or two days ago). Is there a parameter that should be passed into SC2 

Seems like the savgol filter is still in there?

```
pikeSortingError: Spike sorting error trace:
concurrent.futures.process._RemoteTraceback: 
"""
Traceback (most recent call last):
  File "[/usr/lib/python3.8/concurrent/futures/process.py", line 239](http://localhost:8888/usr/lib/python3.8/concurrent/futures/process.py#line=238), in _process_worker
    r = call_item.fn(*call_item.args, **call_item.kwargs)
  File "[/usr/lib/python3.8/concurrent/futures/process.py", line 198](http://localhost:8888/usr/lib/python3.8/concurrent/futures/process.py#line=197), in _process_chunk
    return [fn(*args) for args in chunk]
  File "[/usr/lib/python3.8/concurrent/futures/process.py", line 198](http://localhost:8888/usr/lib/python3.8/concurrent/futures/process.py#line=197), in <listcomp>
    return [fn(*args) for args in chunk]
  File "/home/mxwbio/.local/lib/python3.8/site-packages/spikeinterface/core/job_tools.py", line 439, in function_wrapper
    return _func(segment_index, start_frame, end_frame, _worker_ctx)
  File "/home/mxwbio/.local/lib/python3.8/site-packages/spikeinterface/core/node_pipeline.py", line 522, in _compute_peak_pipeline_chunk
    node_output = node.compute(traces_chunk, *node_input_args)
  File "/home/mxwbio/.local/lib/python3.8/site-packages/spikeinterface/sortingcomponents/waveforms/savgol_denoiser.py", line 59, in compute
    denoised_waveforms = scipy.signal.savgol_filter(waveforms, self.window_length, self.order, axis=1)
  File "[/usr/local/lib/python3.8/dist-packages/scipy/signal/_savitzky_golay.py", line 341](http://localhost:8888/usr/local/lib/python3.8/dist-packages/scipy/signal/_savitzky_golay.py#line=340), in savgol_filter
    coeffs = savgol_coeffs(window_length, polyorder, deriv=deriv, delta=delta)
  File "[/usr/local/lib/python3.8/dist-packages/scipy/signal/_savitzky_golay.py", line 101](http://localhost:8888/usr/local/lib/python3.8/dist-packages/scipy/signal/_savitzky_golay.py#line=100), in savgol_coeffs
    raise ValueError("polyorder must be less than window_length.")
ValueError: polyorder must be less than window_length.
"""
```

zm711, 2024-03-22:
@wesleypclawson,

Is there a reason you want to use python3.8 for this (it's getting close to end of life so to future-proof a bit you may want to change)? I don't think that it will deal with the underlying issue. For that we have to wait for Sam and Pierre to weigh in. We can ping them again early next week.

wesleypclawson, 2024-03-22:
@zm711 Not particularly - I was being a bit lazy and have everything running on the base install, not on conda or a venv, and this is the distribution that was shipped with the computer from Maxwell bio.

yger, 2024-03-22:
This savgol error is strange. What is your sampling rate, and/or the duration of the waveforms you are trying to extract? In theory, the params here should be rather safe, but you can tweak them via some arguments in run_sorter(). Just add clustering = {"smoothing_kwargs": {"window_length_ms": 0.25, "order" : 3}} and hopefully it will work. I'll test more next Monday. If you decrease the order, then it should work. I might add one option to turn that filtering off, I'll keep you updated

yger, 2024-03-22:
And good catch for the np.prod bug :+1:  Thanks everybody!

wesleypclawson, 2024-03-22:
@yger - thanks for the quick reply! the sampling rate is 10k Hz, which may be part of the problem. 

 I added the clustering line you suggested, but the error remains. Happy to test more things if that helps on Monday. 

```
sorting = ss.run_sorter(sorter_name='spykingcircus2',
    recording=recording_preprocessed,
    output_folder="Sorting/spykingcircus2",
    remove_existing_folder=True,
    clustering = {"smoothing_kwargs": {"window_length_ms": 0.25, "order" : 3}},
    verbose=True)
```

Output of spikeinterface_log.json:
```
{
    "sorter_name": "spykingcircus2",
    "sorter_version": "2.0",
    "datetime": "2024-03-22T18:10:45.111785",
    "runtime_trace": [],
    "error": true,
    "error_trace": "concurrent.futures.process._RemoteTraceback: \n\"\"\"\nTraceback (most recent call last):\n  File \"/usr/lib/python3.8/concurrent/futures/process.py\", line 239, in _process_worker\n    r = call_item.fn(*call_item.args, **call_item.kwargs)\n  File \"/usr/lib/python3.8/concurrent/futures/process.py\", line 198, in _process_chunk\n    return [fn(*args) for args in chunk]\n  File \"/usr/lib/python3.8/concurrent/futures/process.py\", line 198, in <listcomp>\n    return [fn(*args) for args in chunk]\n  File \"/home/mxwbio/.local/lib/python3.8/site-packages/spikeinterface/core/job_tools.py\", line 439, in function_wrapper\n    return _func(segment_index, start_frame, end_frame, _worker_ctx)\n  File \"/home/mxwbio/.local/lib/python3.8/site-packages/spikeinterface/core/node_pipeline.py\", line 522, in _compute_peak_pipeline_chunk\n    node_output = node.compute(traces_chunk, *node_input_args)\n  File \"/home/mxwbio/.local/lib/python3.8/site-packages/spikeinterface/sortingcomponents/waveforms/savgol_denoiser.py\", line 59, in compute\n    denoised_waveforms = scipy.signal.savgol_filter(waveforms, self.window_length, self.order, axis=1)\n  File \"/usr/local/lib/python3.8/dist-packages/scipy/signal/_savitzky_golay.py\", line 341, in savgol_filter\n    coeffs = savgol_coeffs(window_length, polyorder, deriv=deriv, delta=delta)\n  File \"/usr/local/lib/python3.8/dist-packages/scipy/signal/_savitzky_golay.py\", line 101, in savgol_coeffs\n    raise ValueError(\"polyorder must be less than window_length.\")\nValueError: polyorder must be less than window_length.\n\"\"\"\n\nThe above exception was the direct cause of the following exception:\n\nTraceback (most recent call last):\n  File \"/home/mxwbio/.local/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py\", line 258, in run_from_folder\n    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)\n  File \"/home/mxwbio/.local/lib/python3.8/site-packages/spikeinterface/sorters/internal/spyking_circus2.py\", line 181, in _run_from_folder\n    labels, peak_labels = find_cluster_from_peaks(\n  File \"/home/mxwbio/.local/lib/python3.8/site-packages/spikeinterface/sortingcomponents/clustering/main.py\", line 44, in find_cluster_from_peaks\n    labels, peak_labels = method_class.main_function(recording, peaks, params)\n  File \"/home/mxwbio/.local/lib/python3.8/site-packages/spikeinterface/sortingcomponents/clustering/random_projections.py\", line 124, in main_function\n    hdbscan_data = run_node_pipeline(\n  File \"/home/mxwbio/.local/lib/python3.8/site-packages/spikeinterface/core/node_pipeline.py\", line 471, in run_node_pipeline\n    processor.run()\n  File \"/home/mxwbio/.local/lib/python3.8/site-packages/spikeinterface/core/job_tools.py\", line 401, in run\n    for res in results:\n  File \"/usr/local/lib/python3.8/dist-packages/tqdm/notebook.py\", line 249, in __iter__\n    for obj in it:\n  File \"/usr/local/lib/python3.8/dist-packages/tqdm/std.py\", line 1182, in __iter__\n    for obj in iterable:\n  File \"/usr/lib/python3.8/concurrent/futures/process.py\", line 484, in _chain_from_iterable_of_lists\n    for element in iterable:\n  File \"/usr/lib/python3.8/concurrent/futures/_base.py\", line 619, in result_iterator\n    yield fs.pop().result()\n  File \"/usr/lib/python3.8/concurrent/futures/_base.py\", line 444, in result\n    return self.__get_result()\n  File \"/usr/lib/python3.8/concurrent/futures/_base.py\", line 389, in __get_result\n    raise self._exception\nValueError: polyorder must be less than window_length.\n",
    "run_time": null
}
```

wesleypclawson, 2024-03-22:
Decreasing order to 1 ->  `clustering = {"smoothing_kwargs": {"window_length_ms": 0.25, "order" : 1}}` let it pass the feature extraction step.

zm711, 2024-03-22:
@wesleypclawson,

If I remember correctly that was an sklearn bug that was affecting hdbscan a few versions ago. #1924. In current versions we don't have this problem. So this one might actually be fixed by updating (I would say 3.10 is the sweet spot since our main tests are on 3.10, but I'm using 3.11 with no problems).

GabrielGaugain, 2024-03-25:
Thanks for the patch @h-mayorquin ! @yger, is there any doc where we could get all the possible param (in spikeinterface API, there aren't the details about included sorters suche as tridesclous2 or spykingcircus) ?

GabrielGaugain, 2024-03-25:
@yger, thank you so much for your feedback. I got the same behavior than @wesleypclawson, same error with win=0.25 and order 2 or 3 but order 1 works. So far, it is running the find spike (with svd) step.

wesleypclawson, 2024-03-25:
Was swamped today - but tomorrow (or Wednesday) will update to 3.10 and try re-running and will post results.

zm711, 2024-03-30:
@GabrielGaugain, we actually do have two functions that help with seeing params and meanings.

`get_default_sorter_params`: returns the default params. 
(for example `ss.get_default_sorter_params('spykingcircus2')`)

and `get_sorter_params_description`: returns the documentation on what the parameters mean 
(for example `ss.get_sorter_params_description('spykingcircus2')`.

zm711, 2024-12-03:
I think we have finally worked on this. If others run into these issues on Windows please open a new issue!

closed 2024-12-03 by zm711