SpikeInterface/spikeinterface #1876: [Bug] Intan not detecting digital stream

CodyCBakerPhD, 2023-07-25:
stems from https://github.com/catalystneuro/neuroconv/issues/519

The header of the IntanRecordingExtractor is not detecting a digital channel from the .rhd file

@jeongjunjjkim may have to share the example file with whoever needs to fix it here

Likely needs to be passed up to neo, but wanted to ask here first

CodyCBakerPhD, 2023-07-30:
@zm711 Thank you so much for the info, will keep an eye on it

zm711, 2023-10-06:
@CodyCBakerPhD Do you want to leave this open until a new official Neo release? I'm just going through some of the older issues to make sure they stay on people's radar.

CodyCBakerPhD, 2023-10-06:
> @CodyCBakerPhD Do you want to leave this open until a new official Neo release? I'm just going through some of the older issues to make sure they stay on people's radar.

Thanks for checking in! I do that myself, too

I suppose I'd be fine with closing since it can now be done at the NeuroConv level even if we can't cut an official release until such time as neo does

closed 2023-10-06 by CodyCBakerPhD