SpikeInterface/spikeinterface #1877: working_with_tetrodes.py is an empty file

JohnStout, 2023-07-25:
Title says it all!

h-mayorquin, 2023-07-26:
If I read well this was intended to be a documentation file. Is that what you were looking for? Documentation on how to work with tetrodes? If so, anything specific you will ike to see there?

This type of feedback will be useful for us.

alejoe91, 2023-07-26:
Yep that's a place holder! @JohnStout do you have by any chance a tetrode recording you'd be willing to share? We mainly work with other probes so it would be great to use some recent tetrodes data!

JohnStout, 2023-08-03:
Yes, I’ll send this today!

JohnStout, 2023-08-03:
Hey Alessio,

I've attached different datasets for different purposes. All acquired with
Neuralynx and processed on Cheetah.

First, I've attached an NWB file with unfiltered recordings from 4
tetrodes. I was only able to extract 4 units with spikeinterface
(tridesclous), which seems strange because that would be one unit/tetrode
(I would expect an uneven distribution of units per tetrode). In this NWB
file, there are 4 CSCs (LFPs) and 16 channels grouped into 4 tetrodes...
I've also attached a script (see lines 51:end) to see what I did
("nwbSpikeInterface.py").

Second, here is a link to a dataset with head-tracking data (LEDs mounted
to a recording headstage) from a rat performing a T-maze task. I'm trying
to find a way to extract behavior, as was monitored using LED's mounted to
the rats headstage (see VT1.nvt). Specifically, the .nvt file has X/Y and
Timestamp data (sampled at ~29-30samples/sec), synchronized with the LFP
times. There may also exist an "angles" variable, but this is less
important (for my data at least). Likewise, neuralynx has an "Events.nev"
file which contains timestamps of interest, but I think spikeinterface has
a way to handle this? Here is the link to the google drive:
https://drive.google.com/drive/folders/1zLTovUMjh5S7RqPh1-JE8RlayMXsPYDJ?usp=share_link

Thank you!
John
 data_nwb
<https://drive.google.com/file/d/14lrvEEYErB0AfhjPlK9mjbKXSDfngmDM/view?usp=drive_web>

PR #4055 (2025-07-28): Add working with tetrodes doc
Add PR giving advice about working with tetrodes. Contains
- Adding a 2-tetrode ProbeGroup and attaching to the recording (not in place!!!!)
- Removing bad channels
- Preprocessing by group

Docs: https://spikeinterface--4055.org.readthedocs.build/en/4055/tutorials/forhowto/plot_1_working_with_tetrodes.html

Also the first example of a How To guide that we create as a sphnx-gallery item (but include in the How Tos rather than the Tutorials). Is pretty easy to do!

Closes #1877