SpikeInterface/spikeinterface #1880: How to adjust sorter parameters for klustakwik?

SusanL82, 2023-07-26:
Hello, 

I finally managed to get my clustering with klustakwik working (yay!), but the whole process seems incredibly slow. 

The entire clustering procedure for a single tetrode in a ~3hour recording took over 5 hours (1 hour for spike detection and feature computing). 
This seems like a lot, because running klustakwik via different software (i.e. 
[klusta.log](https://github.com/SpikeInterface/spikeinterface/files/12173480/klusta.log)
NLX's SpikeSort3D) takes about half the time with mostly the same settings. 
One big difference is that we generally set the maximum number of detected clusters to 100 per tetrode (the default is 1000).  I think that this could explain the speed difference. 
Is there some way to adjust klustakwik.initial_parameters max_possible_clusters? It is not available as a parameter in run_sorter(). 

Susan
[klusta.log](https://github.com/SpikeInterface/spikeinterface/files/12173547/klusta.log)

zm711, 2024-01-11:
@alejoe91, did you want to leave this open? @SusanL82, did you want to pursue Alessio's suggestion or have you decided to do something else?

zm711, 2024-03-25:
I'll close this, but reopen/open a new issue if you have more questions @SusanL82 :)

closed 2024-03-25 by zm711