SpikeInterface/spikeinterface #1882: `TdtRecordingExtractor` not reading signal streams correctly for multi-block experiment

weiglszonja, 2023-07-26:
This experiment has 3 signal streams with mult-block (10 blocks) setup, the reader `TdtRawIO` can parse the header but `TdtRecordingExtractor` returns empty list of times and traces, sampling frequency is `1`, the data type is `int`. 
The data for one of the blocks look like https://gin.g-node.org/NeuralEnsemble/ephy_testing_data/src/master/tdt/aep_05/Block-1 but also have multiple SEV files like in https://gin.g-node.org/NeuralEnsemble/ephy_testing_data/src/master/tdt/dataset_1_single_block. 

@JuliaSprenger have you seen similar before with multiple data streams and multiple block example?

JuliaSprenger, 2023-08-03:
Hi @weiglszonja,
yes, I have seen this with some other datasets already. The base of the issue is that the TdT format comes in many different flavours and it seems not all streams are picked up correctly yet for all of them. I guess your issue is the same as https://github.com/NeuralEnsemble/python-neo/issues/1286 ?

M-Hisham, 2025-03-13:
Hello, Thank you very much for the great work

I have similar issue. We are working on 128 channels, and each trial contains 128 sev files that are not loaded. it shows the correct number of channels, but the type, samples and size it not correct, and simply empty.
I am using the function by passing one file inside the folder. it can load all data for streams, except the one stream saved in the sev files.

[image].

[image]

The documentation for the TDT python package can read it correctly
https://www.tdt.com/docs/sdk/offline-data-analysis/offline-data-python/

is there a way to pass loaded data to spikeInterface directly?
I have been looking at issue [#2897](https://github.com/SpikeInterface/spikeinterface/issues/2897#issue-2311562644), but I couldn't resolve the problem.

I appreciate your support

zm711, 2025-03-18:
@M-Hisham what does the data structure look like? Does the tdt package give you the option to write to another format (e.g. openephys?)

M-Hisham, 2025-03-19:
@zm711 
Thanks for your help
I have uploaded a trial for my data in the original TDT format [here](https://tuprd-my.sharepoint.com/:f:/g/personal/tuu21853_temple_edu/EolXnZh8TR5Hhczt3LtOQWYBDpniLAvRrbo7uwEKsG8IzQ?e=FAvY4L)

the sev files stream is named as "RSn1"

The main data we are interested in is in the SEV files, each file contain the data stream for one channel.
The folder size is around 1GB, sorry for that, but I wasn't sure if loading the data correctly would still work with less channels.