SpikeInterface/spikeinterface #1885: Error during filtering for TracePaddedRecording

TomBugnon, 2023-07-26:
And here's the next one !

```
from spikeinterface.full import NumpyRecording, concatenate_recordings, write_binary_recording, highpass_filter, phase_shift, bandpass_filter
from spikeinterface.preprocessing.zero_channel_pad import TracePaddedRecording

sampling_frequency = 30000
base_duration = 99.000
base_nsamp = int(base_duration * sampling_frequency)
base_traces = np.zeros((base_nsamp, 5), dtype="float64")

rec = NumpyRecording([base_traces], sampling_frequency)
# NumpyRecording: 5 channels - 30.0kHz - 1 segments - 2,970,000 samples - 99.00s (1.65 minutes) 
#                float64 dtype - 113.30 MiB

prepro_rec = bandpass_filter(rec, 300, 12000)
# BandpassFilterRecording: 5 channels - 30.0kHz - 1 segments - 2,970,000 samples 
#                         99.00s (1.65 minutes) - float64 dtype - 113.30 MiB

# Actual number of padded samples when running kilosort on a recording with this length
prepro_padded_rec = TracePaddedRecording(prepro_rec, 0, 47600)
# TracePaddedRecording: 5 channels - 30.0kHz - 1 segments - 3,017,600 samples 
#                      100.59s (1.68 minutes) - float64 dtype - 115.11 MiB

# When trying to write the last chunk in write_binary_recording
prepro_padded_rec.get_traces(0, 3000000, 3017600)
```

> ValueError                                Traceback (most recent call last)
> Cell In[62], line 2
>       1 # When trying to write the last chunk in write_binary_recording
> ----> 2 prepro_padded_rec.get_traces(0, 3000000, 3017600)
> 
> File ~/projects/ecephys_dev/spikeinterface/src/spikeinterface/core/baserecording.py:278, in BaseRecording.get_traces(self, segment_index, start_frame, end_frame, channel_ids, order, return_scaled, cast_unsigned)
>     276 channel_indices = self.ids_to_indices(channel_ids, prefer_slice=True)
>     277 rs = self._recording_segments[segment_index]
> --> 278 traces = rs.get_traces(start_frame=start_frame, end_frame=end_frame, channel_indices=channel_indices)
>     279 if order is not None:
>     280     assert order in ["C", "F"]
> 
> File ~/projects/ecephys_dev/spikeinterface/src/spikeinterface/preprocessing/zero_channel_pad.py:101, in TracePaddedRecordingSegment.get_traces(self, start_frame, end_frame, channel_indices)
>      99 # After the padding, the original traces are placed in the middle until the end of the original traces
>     100 if end_frame >= self.padding_start:
> --> 101     original_traces = self.get_original_traces_shifted(
>     102         start_frame=start_frame,
>     103         end_frame=end_frame,
>     104         channel_indices=channel_indices,
>     105     )
>     107     padded_frames_at_the_start = max(self.padding_start - start_frame, 0)
>     108     end_of_original_traces = padded_frames_at_the_start + original_traces.shape[0]
> 
> File ~/projects/ecephys_dev/spikeinterface/src/spikeinterface/preprocessing/zero_channel_pad.py:125, in TracePaddedRecordingSegment.get_original_traces_shifted(self, start_frame, end_frame, channel_indices)
>     123 original_end_frame = min(end_frame - self.padding_start, self.num_samples_in_original_segment)
>     124 print(original_start_frame, original_end_frame)
> --> 125 original_traces = self.parent_recording_segment.get_traces(
>     126     start_frame=original_start_frame,
>     127     end_frame=original_end_frame,
>     128     channel_indices=channel_indices,
>     129 )
>     131 return original_traces
> 
> File ~/projects/ecephys_dev/spikeinterface/src/spikeinterface/preprocessing/filter.py:148, in FilterRecordingSegment.get_traces(self, start_frame, end_frame, channel_indices)
>     145 import scipy.signal
>     147 if self.filter_mode == "sos":
> --> 148     filtered_traces = scipy.signal.sosfiltfilt(self.coeff, traces_chunk, axis=0)
>     149 elif self.filter_mode == "ba":
>     150     b, a = self.coeff
> 
> File ~/miniconda3/envs/ecephys_dev/lib/python3.10/site-packages/scipy/signal/_signaltools.py:4411, in sosfiltfilt(sos, x, axis, padtype, padlen)
>    4409 ntaps = 2 * n_sections + 1
>    4410 ntaps -= min((sos[:, 2] == 0).sum(), (sos[:, 5] == 0).sum())
> -> 4411 edge, ext = _validate_pad(padtype, padlen, x, axis,
>    4412                           ntaps=ntaps)
>    4414 # These steps follow the same form as filtfilt with modifications
>    4415 zi = sosfilt_zi(sos)  # shape (n_sections, 2) --> (n_sections, ..., 2, ...)
> 
> File ~/miniconda3/envs/ecephys_dev/lib/python3.10/site-packages/scipy/signal/_signaltools.py:4180, in _validate_pad(padtype, padlen, x, axis, ntaps)
>    4178 # x's 'axis' dimension must be bigger than edge.
>    4179 if x.shape[axis] <= edge:
> -> 4180     raise ValueError("The length of the input vector x must be greater "
>    4181                      "than padlen, which is %d." % edge)
>    4183 if padtype is not None and edge > 0:
>    4184     # Make an extension of length `edge` at each
>    4185     # end of the input array.
>    4186     if padtype == 'even':
> 
> ValueError: The length of the input vector x must be greater than padlen, which is 33.

TomBugnon, 2023-07-26:
I think now it is FilterRecordingSegment that doesn't handle prpoerly `start_frame > num_samples` or `start_frame > end_frame`

samuelgarcia, 2023-07-27:
I think that we should make the API harder and raise error to prevent this kind of weird call with a checker in the base function.
start_frame > num_samples should not be allowed
and 
start_farme < stop_frame should not be allowed

h-mayorquin, 2024-06-20:
Should have beeen closed with https://github.com/SpikeInterface/spikeinterface/pull/1883

closed 2024-06-20 by h-mayorquin