SpikeInterface/spikeinterface #1889: Questions about the modes "median"/"average" in `remove_artifacts`

oaaij-gnahz, 2023-07-26:
Hi!

I am trying to explore the functionality of [`remove_artifacts`](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/preprocessing/remove_artifacts.py) with mode "median" or "average" and have a few questions:

(1) Just to confirm if my understanding is correct or not: The input argument `artifacts` gives a set of templates that suggest what each artifact looks like; if `None`, then the artifacts are calculated from the raw traces by indexing the triggers and extracting "waveforms". Additionally the "jitter" parameter is used to find a best match between the recording traces and the templates and the corresponding amplitude scaling factor. Is it the right way to interpret it? The [documentation](https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.preprocessing.remove_artifacts) only says that they are the "values" of the artifacts. I wonder in which cases the artifacts need to be explicitly specified; when specified, it is assumed that all the channels have the same artifact waveform?

(2) The documentation says that providing argument `artifacts` can bypass `ms_before` and `ms_after`. But in the implementation of [`get_traces`](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/preprocessing/remove_artifacts.py#L417) in `RemoveArtifactsRecordingSegment`, in "median"/"average" mode, the slicing of traces still depend on the `pad` property, which is derived from `ms_before` and `ms_after`. Then it seems to me that we need to manually make sure arguments `ms_before` and `ms_after` need to be aligned with `artifacts` so that they indicate the same length. Otherwise there may be mismatch when subtracting the artifacts from the recording traces. I am not sure if I understood the logic in source code correctly, nor have I tested.

(3) For my own use case, I wanted to mask my stimulation artifacts to zero. But different stimulation artifacts have different durations. It seems like merely using `zeros` mode wouldn't accommodate different artifacts. If I use `median`/`average` mode while providing labels, it will calculate the artifact waveform template by the label and the subtract from raw data, not quite what I wanted to do. Could you please suggest some efficient ways to zero out the artifacts with varying durations?

Thank you so much for your help. I really appreciate your active maintenance and development of this framework. It has been making my life a lot easier!

yger, 2023-07-27:
(1) The idea behind the function is that indeed, ideally you should be able to provide a list of (time, ids) for your template (with time scales ms_before and ms_after). If you do not provide any artefacts, then it will be computed on the fly based on your times/ids, and you will get one dictionary (with a waveformextractor object). Once you have this dictionary, indeed, you have several options:
- either you simply subtract, at the appropriate time, the mean/median templates.
- either you allow for some distortions (in time with temporal jittering, or in amplitudes with the scaling). 

(2) For the pad, I'll have a look. Indeed, the intented behavior is that if you provide ms_before/ms_after for zeroing or interpolating at the times of your artefacts, you want these values to be used. If, on the other hand, you are providing your own dictionary of templates that can have been precomputed, then the ms_before and ms_after values of this dictionary should be used. But to be honest, the reasons why artefacts can be provided is mostly to allow such a function to be serialized and launched on several cores with the internal machinery of spike interfaces. This should not really be a proper use case, and in practise ms_before and ms_after should be the same as the ones in the dict.

(3) The objects you should look for is SilencePeriods https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/preprocessing/silence_periods.py
With such an object, you can provide a list of t_start, t_stop, and signal will be zero in between on all these periods. Note, however, that you can not specify a subset of channels per periods, but this is not what you wan't to do, is it? This might be a natural extension for such an object

oaaij-gnahz, 2023-07-27:
Got it. SliencePeriods seems to be exactly what I need and I mean to zero all channels. Thank you!

closed 2023-07-27 by oaaij-gnahz