SpikeInterface/spikeinterface #1890: Using spikeinterface with a spike sorter installed in a different environment

grg2rsr, 2023-07-27:
Hi,
I have a simple question regarding the installation of spike sorters that I can't seem to find an answer for in th docs. I have pykilosort previously installed in a different environment, and now looking into spikeinterface. Managing the CUDA mess was nothing that I necessarily want to go through again. Can I somehow tell spikeinterface where to find the sorter, as in tell the wrapper the environment of installation or something similar? If I understood [the docs](https://spikeinterface.readthedocs.io/en/latest/modules/sorters.html#external-sorters-the-wrapper-concept) correctly, the sorters are started as a different process in any case. I was hoping that something like this exists:

```
sorting_pyks = ss.run_sorter("pykilosort", recording, output_folder="spikeinterface_pyks", env_name= "pyks2")
```
but doesn't seem like.
Any chance I missed something? Thanks

alejoe91, 2023-07-27:
Hi @grg2rsr 

Unfortunately that is not an option. The sorters are started in the same process when run locally, so selecting an a different environment would require some changes and some additional flexibility (why not supporting venv too?)

Have you looked into using docker/singularity? That should solve any installation issue, since the sorter doesn't need to be installed.

Cheers
Alessio

grg2rsr, 2023-07-27:
I struggled with getting docker to work in the first place (the installation instructions linked in the docs were not working for me). I managed to push through those now, at least the docker hello-world run.

Spikeinterface gives me the following error now

```
Docker: pulling image spikeinterface[/pykilosort-base](https://file+.vscode-resource.vscode-cdn.net/pykilosort-base)
---------------------------------------------------------------------------
HTTPError                                 Traceback (most recent call last)
File [~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/docker/api/client.py:268](https://file+.vscode-resource.vscode-cdn.net/home/georg/Projects/FreelyMoving/~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/docker/api/client.py:268), in APIClient._raise_for_status(self, response)
    [267](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/docker/api/client.py?line=266) try:
--> [268](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/docker/api/client.py?line=267)     response.raise_for_status()
    [269](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/docker/api/client.py?line=268) except requests.exceptions.HTTPError as e:

File [~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/requests/models.py:1021](https://file+.vscode-resource.vscode-cdn.net/home/georg/Projects/FreelyMoving/~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/requests/models.py:1021), in Response.raise_for_status(self)
   [1020](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/requests/models.py?line=1019) if http_error_msg:
-> [1021](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/requests/models.py?line=1020)     raise HTTPError(http_error_msg, response=self)

HTTPError: 400 Client Error: Bad Request for url: http+docker://localhost/v1.41/containers/19e8c77108a977b3b064e936e97a6e858e30c945837d4681bcd088c5000337fa/start

The above exception was the direct cause of the following exception:

APIError                                  Traceback (most recent call last)
[/home/georg/Projects/FreelyMoving/spikeinterface_testing.py](https://file+.vscode-resource.vscode-cdn.net/home/georg/Projects/FreelyMoving/spikeinterface_testing.py) in line 2
      [36](file:///home/georg/Projects/FreelyMoving/spikeinterface_testing.py?line=35) # %%
----> [37](file:///home/georg/Projects/FreelyMoving/spikeinterface_testing.py?line=36) sorting_KS2_5 = ss.run_sorter("pykilosort", recording, output_folder="[/home/georg/data_local/D1Chr2_6351_240723/spikeinterface_ks25/](https://file+.vscode-resource.vscode-cdn.net/home/georg/data_local/D1Chr2_6351_240723/spikeinterface_ks25/)", docker_image=True)

File [~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:142](https://file+.vscode-resource.vscode-cdn.net/home/georg/Projects/FreelyMoving/~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:142), in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, **sorter_params)
    [140](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=139)         else:
    [141](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=140)             container_image = singularity_image
--> [142](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=141)     return run_sorter_container(
    [143](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=142)         container_image=container_image,
    [144](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=143)         mode=mode,
    [145](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=144)         **common_kwargs,
    [146](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=145)     )
    [148](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=147) return run_sorter_local(**common_kwargs)

File [~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:489](https://file+.vscode-resource.vscode-cdn.net/home/georg/Projects/FreelyMoving/~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:489), in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, delete_container_files, extra_requirements, **sorter_params)
    [487](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=486) if verbose:
    [488](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=487)     print("Starting container")
[... cut ...]
APIError: 400 Client Error for http+docker://localhost/v1.41/containers/19e8c77108a977b3b064e936e97a6e858e30c945837d4681bcd088c5000337fa/start: Bad Request ("failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error running hook #0: error running hook: exit status 1, stdout: , stderr: Auto-detected mode as 'legacy'
nvidia-container-cli: initialization error: load library failed: libnvidia-ml.so.1: cannot open shared object file: no such file or directory: unknown")
```

grg2rsr, 2023-07-27:
after some lost and won battles with docker, I am getting the following error from spikeinterface
```
---------------------------------------------------------------------------
HTTPError                                 Traceback (most recent call last)
File [~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/docker/api/client.py:268](https://file+.vscode-resource.vscode-cdn.net/home/georg/Projects/FreelyMoving/~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/docker/api/client.py:268), in APIClient._raise_for_status(self, response)
    [267](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/docker/api/client.py?line=266) try:
--> [268](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/docker/api/client.py?line=267)     response.raise_for_status()
    [269](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/docker/api/client.py?line=268) except requests.exceptions.HTTPError as e:

File [~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/requests/models.py:1021](https://file+.vscode-resource.vscode-cdn.net/home/georg/Projects/FreelyMoving/~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/requests/models.py:1021), in Response.raise_for_status(self)
   [1020](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/requests/models.py?line=1019) if http_error_msg:
-> [1021](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/requests/models.py?line=1020)     raise HTTPError(http_error_msg, response=self)

HTTPError: 400 Client Error: Bad Request for url: http+docker://localhost/v1.41/containers/17839a28557d56af412be33ba18377bf1e9e711fd6beefe4b751d770438d9906/start

The above exception was the direct cause of the following exception:

APIError                                  Traceback (most recent call last)
[/home/georg/Projects/FreelyMoving/spikeinterface_testing.py](https://file+.vscode-resource.vscode-cdn.net/home/georg/Projects/FreelyMoving/spikeinterface_testing.py) in line 2
      [36](file:///home/georg/Projects/FreelyMoving/spikeinterface_testing.py?line=35) # %%
----> [37](file:///home/georg/Projects/FreelyMoving/spikeinterface_testing.py?line=36) sorting_KS2_5 = ss.run_sorter("pykilosort", recording, output_folder="[/home/georg/data_local/D1Chr2_6351_240723/spikeinterface_ks25/](https://file+.vscode-resource.vscode-cdn.net/home/georg/data_local/D1Chr2_6351_240723/spikeinterface_ks25/)", docker_image=True)

File [~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:142](https://file+.vscode-resource.vscode-cdn.net/home/georg/Projects/FreelyMoving/~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:142), in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, **sorter_params)
    [140](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=139)         else:
    [141](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=140)             container_image = singularity_image
--> [142](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=141)     return run_sorter_container(
    [143](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=142)         container_image=container_image,
    [144](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=143)         mode=mode,
    [145](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=144)         **common_kwargs,
    [146](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=145)     )
    [148](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=147) return run_sorter_local(**common_kwargs)

File [~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:489](https://file+.vscode-resource.vscode-cdn.net/home/georg/Projects/FreelyMoving/~/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:489), in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, delete_container_files, extra_requirements, **sorter_params)
    [487](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=486) if verbose:
    [488](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=487)     print("Starting container")
--> [489](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=488) container_client.start()
    [491](file:///home/georg/anaconda3/envs/spikeinterface/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py?line=490) if verbose and install_si_from_source:
[... cut ...]
APIError: 400 Client Error for http+docker://localhost/v1.41/containers/17839a28557d56af412be33ba18377bf1e9e711fd6beefe4b751d770438d9906/start: Bad Request ("failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error running hook #0: error running hook: exit status 1, stdout: , stderr: Auto-detected mode as 'legacy'
nvidia-container-cli: initialization error: load library failed: libnvidia-ml.so.1: cannot open shared object file: no such file or directory: unknown")
```
I have all nvidia related things installed, including the nvidia-container-toolkit

alejoe91, 2023-07-27:
What OS are you using? Are you able to run other sorters via Docker? Does this command work?
```
docker run -it --gpus all spikeinterface/pykilosort-base:latest
```

And inside the container run: `nvidia-smi`?

Have you followed all steps here (https://spikeinterface.readthedocs.io/en/latest/modules/sorters.html#containerizedsorters)  to set up docker with the GPU?

grg2rsr, 2023-07-27:
thanks for your response!
```
docker: Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error running hook #0: error running hook: exit status 1, stdout: , stderr: Auto-detected mode as 'legacy'
nvidia-container-cli: initialization error: load library failed: libnvidia-ml.so.1: cannot open shared object file: no such file or directory: unknown.
ERRO[0002] error waiting for container: context canceled
```
currently looking into https://github.com/NVIDIA/nvidia-docker/issues/1648 and https://github.com/NVIDIA/nvidia-docker/issues/1711 ... I think probably related.

samuelgarcia, 2023-07-27:
Hi @grg2rsr.
Thanks you for the suggestion.
running a sorter in a differents env should be not so a big patch in our API for python based sorters.

alejoe91, 2023-07-27:
@samuelgarcia we would still need a `run_sorter_in_subprocess()` function and make a script similar to the container one. No?

samuelgarcia, 2023-07-28:
@alejoe91 : I would have in mind more something than we do for matlab prepare a script and run it in a diffrent env like we do for matlab not somethign complicated like docker where SI must be installed in the other envs.