SpikeInterface/spikeinterface #1896: Cleanup of kilosort temporary files causing issues with Phy

JoeZiminski, 2023-07-28:
I was planning to raise an issue on the cleanup of intermediate files in Kilosort but I see this has already been done in #1709 which is great. I would suggest some supplementary additions that allow a little more refinement over the selection of which intermediate files to delete. Currently `temp_wh.dat` is deleted by default and (at least in my hands) this causes Phy to not be able to run as it requires `temp_wh.dat`.

I believe there are three cases of intermediate files created during Kilosort-ing with spikeinterface:
1) `recording.dat` - binary recording copied into 'sorting_output' by spikeinterface (useful if concatenating across segments)
2) `temp_wh.dat` - recording that has been preprocessed (e.g. drift corrected) in Kilosort. Despite the confusing name this is required by Phy for visualisation (by default) unless the corresponding path in `params.py` is changed. But as discussed [here](https://github.com/MouseLand/Kilosort/issues/311), it makes sense to visualise based on the preprocessed data.
3) intermediate matlab files

I would suggest changing the arguments and defaults `delete_tmp_files=True` and `delete_recording_dat=False` to a tuple argument
`delete_intermediate_files: Tuple[Literal["recording.dat", "temp_wh.dat", "matlab_files"]] = ("recording.dat", "matlab_files")`. A benefit of this is it allows full flexibility without adding a third argument (for the matlab files). But maybe there is a good use case for keeping the `recording.dat` by default too and / or this signature is confusing.

alejoe91, 2023-08-01:
Hi @JoeZiminski 

That is a great suggestion! Would you have time to make a PR for this?

Cheers!

JoeZiminski, 2023-08-02:
Great! Sure I will do this

zm711, 2023-08-21:
@JoeZiminski quick question: you linked back to Kilosort where it is explained that the switch over occurred between v2 and v2.5 for Kilosort. For users pre 2.5 (so 2 and 1 if anyone still uses it), I believe the default file should be the `recording.dat` instead of `temp_wh.dat`.  Although they are preprocessed they are not drift corrected and so Matlab itself includes the raw data file rather than the `temp_wh.dat`.  I guess you could make an argument that that was a mistake and that even in old versions the `temp_wh.dat` should be used instead of the raw file. But curious to see what you think.

Your default of deleting neither (`recording.dat` or `temp_wh.dat`) seems sensible (in #1908), but currently if exporting (to Phy) from SpikeInterface rather than from Matlab the only option is `recording.dat` (see code snippet below--although I might be missing other options in that complicated code bit), so part of your issue might also stem from any switching back and forth between Matlab exported "Phy files" and exporting from SpikeInterface itself, which could switch the params.py back to `recording.dat` without any warning, but I guess that depends on each individual user's method of performing Phy with SpikeInterface.

https://github.com/SpikeInterface/spikeinterface/blob/94df77a43532fbe8671026d6aa1eb528c755a5ef/src/spikeinterface/exporters/to_phy.py#L131-L146

Maybe in this case the `to_phy` would also need an update to help with the issue of choosing which binary file to use-- (with a parameter since it needs to handle all the other sorters too? @alejoe91)?

Also possible I'm totally missing something and am off-base, but was just curious.

JoeZiminski, 2023-08-29:
Hi @zm711 thanks for this! I hadn't really thought properly about kilosort 1 and 2. That's very true, I think as you mention in this case then deleting nothing by default is wise, and then either `recording.dat` or `temp_wh.dat` can be deleted by choice depending on the Kilosort version used. My guess is that it would still be correct to use `temp_wh.dat` for visualisation even in Kilosort1 and 2 because of the filtering and whitening operations (discussed more in the final paragraph)..

You might be interested in #1916, here @alejoe91 suggested the idea of a `KilosortTempWhRecording` that handles the problem of removed channels from the `temp_wh.dat` and allow it to be passed to  `extract_waveforms`. Then the waveforms used for spikeinterface quality checks will be read from the kilosort-preprocessed data used for sorting. I think for the `export_to_phy` function, this will create an entirely new output folder that will never contain a `temp_wh.dat` file, and so hopefully there would be no chance for a conflict in this case. As I understand it, for anything sorted in Kilosort this function would not be necessary as the sorting output is already Phy-compatible? Nonetheless, a `KilosorttempWhRecording` object could be passed to `extract_waveforms` instead of the preprocessed SpikeInterface `recording` object, and then `export_to_phy` could be run,  providing a mechanism to choose between the two binaries.

This is a very good point though about kilosort 1 and 2. For kilosort >= 2.5, it is clear that waveforms for quality metrics should be extracted from the `temp_wh.dat`, and a `KilosortTempWhRecording` class will help this. This also ensures that all postprocessing outputs from SpikeInterface will match Phy (e.g. unit locations).

However, in the case of kilosort 1 and 2 it is not 100% clear what should be done, because Phy itself is not using the `recording.dat`. My guess is that it is still more correct to still extract waveforms from `temp_wh.dat` because then the quality metrics are performed on exactly the same data that was used for sorting (i.e. after any kilosort filtering or whitening operations that have been performed). However, the fact that Phy does not use this data could cause some confusion. I think in the case of Kilosort 1 and 2 it would make sense to check with the Kilosort team that indeed it would be better to use the `temp_wh.dat` for calculating all quality metrics. In terms of the data that is loaded into Phy, I wonder if the use of `recording.dat` in Kilosort 1, 2 is not too problematic because most of the data is already written in `.npy` files that were presumably calculated from the kilosort-preprocessed data. Maybe the only thing that would be incorrect is the raw waveforms? but I am not that familiar with Phy or Kilosort's inner workings, though I think in many cases (e.g. `template_features.npy`, `templates.npy`) the actual data is stored, rather than indices that are extracted from whatever data is passed. Anyway, in terms of SpikeInterface waveform extraction and quality-metrics calculation my guess is that it would be better in kilosort versions to use the `temp_wh.dat` file.

zm711, 2023-08-29:
@JoeZiminski 

Great explanation really appreciate the time and thought in that and pretty similar to how I felt. And yeah using the `recording.dat` for 1 and 2 would really only cause the user to lose a high-pass filter and whitening, but the `FeatureView` is based on the `.npy` files which were calculated after that preprocessing so that shouldn't be too big a problem.

Just one additional point, some people may sort with Kilosort.x  (which as you said is Phy compatible), then they curate a little in Phy. Then they use the PhyExtractor to bring back into SpikeInterface and then they might want to reexport to go back for more Phy stuff. So one concern would be making sure the files don't switch out from under people if they run the export to phy as they go back and forth. But I honestly don't know how common that type of user is.

If a user is mainly doing preproc, sort, qc then the `KilosorttempWhRecording` looks perfect though. Thanks again.

PR #1908 (2023-10-17): Change the signature on kilosort's delete intermediate files parameters.
This PR extends cleanup after kilosort sorting, closes #1896.

The signature for deleting intermediate files is changed slightly and is now a `Tuple` with optional entries, `recording.dat`, `temp_wh.dat` and `matlab_files`. The reason for this extension is to give a little more granularity over the selection of what to delete, and to not delete `temp_wh.dat` by default as it is used for Phy.

For now, have default to delete only `matlab_files` because my understanding is that `recording.dat` is not deleted for other sorte [... cut]
closed 2023-10-17 by alejoe91