SpikeInterface/spikeinterface #1897: NWBRecordingExtractor: fix channel IDs reference

CodyCBakerPhD, 2023-07-28:
This line https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/extractors/nwbextractors.py#L213

causes trouble for AP/Lf/multiprobe setups

```python
from nwbinspector.tools import get_s3_urls_and_dandi_paths
from pynwb import NWBHDF5IO
from spikeinterface.extractors import NwbRecordingExtractor

s3_urls = get_s3_urls_and_dandi_paths(dandiset_id="000409")
dandi_paths = {v: k for k, v in s3_urls.items()}

s3_url = dandi_paths["sub-CSH-ZAD-001/sub-CSH-ZAD-001_ses-3e7ae7c0-fe8b-487c-9354-036236fa1010_behavior+ecephys+image.nwb"]

recording = NwbRecordingExtractor(file_path=s3_url, stream_mode="ros3", electrical_series_name="ElectricalSeriesAp")

num_channels = recording.get_num_channels()

num_channels
> 768

> recording.get_traces(start_frame=0, end_frame=2).shape
(2, 384)
```

The actual `get_traces` call is trustworthy and working as intended, but the proper channel ID fetching could be done better through the dynamic table region reference

alejoe91, 2023-08-01:
Thanks @CodyCBakerPhD 

What I would do is to check the shape of the electrical series first. If it also has a dynamic table reference to the electrodes table, then use it. Otherwise on case of a mismatch use the first N electrodes and raise a warning. Sounds fair?

CodyCBakerPhD, 2023-08-01:
@alejoe91 Yep, sounds good to me

CodyCBakerPhD, 2023-08-02:
Can confirm - latest SI main does not seem to have this issue

Given the timing of the last release and the blame for that line, I'm not sure what else would have changed...

Feel free to close this as a non-issue then?

alejoe91, 2023-08-02:
I think it was probably a version before that! Good that w ehave checked!

closed 2023-08-02 by alejoe91