SpikeInterface/spikeinterface #1899: plot_traces error in neuropixel guide

nmtimme, 2023-07-31:
Hello

I'm trying to go through the guide to using spike interface to spike sort neuropixel data (https://spikeinterface.readthedocs.io/en/latest/how_to/analyse_neuropixels.html), but I'm running into an error with plot_traces not being an attribute of spikeinterface.

I'm running version 0.98.2 on a Linux cluster with a GPU. I'm running on a virtual desktop, so I can view plots and interact with GUIs. I installed spikeinterface using miniconda 4.12.0 using this .yml file:

```
name: si_env
channels:
  - conda-forge
  - defaults
dependencies:
  - python=3.9
  - pip>=21.0
  - mamba
  # numpy 1.22 break numba which break tridesclous
  - numpy<1.22
  # joblib 1.2 is breaking hdbscan
  - joblib=1.1
  - tqdm
  - matplotlib
  - h5py
  - pandas
  - scikit-learn
  - networkx
  - pybind11
  - loky
  - hdbscan
  - numba
  - jupyter
  - mpi4py
  - cxx-compiler
  - libxcb
  - pyqt=5
  - pyqtgraph
  - htop
  - ipywidgets
  - ipympl
  - pip:
    - ephyviewer
    - neo>=0.11
    - elephant>=0.10.0
    - probeinterface>=0.2.11
    - MEArec>=1.8
    - spikeinterface[full, widgets]
    - spikeinterface-gui
    - tridesclous>=1.6.6.1
    - spyking-circus>=1.1.0
    # - phy==2.0b5
```

I am running the following script when I encounter the error:

```
import spikeinterface.full as si
print(si.__version__)

import numpy as np
import matplotlib.pyplot as plt
from pathlib import Path

# Define the path to the data
base_folder = Path('/N/project/lapishLabWorkspace/NickPilotNeuropixelEphysData')
openephys_folder = base_folder / '2023-03-15_15-18-59'

# Figure out the data streams in this data set
stream_names, stream_ids = si.get_neo_streams('openephys', openephys_folder)
print('stream_names',stream_names)

# Read the data LFP data from probe A (this automatically loads the probe map)
raw_rec = si.read_openephys(openephys_folder, stream_name='Record Node 101#Neuropix-PXI-100.ProbeA-AP')
print('raw_rec',raw_rec)

# Plot the bottom part of the channel map
fig, ax = plt.subplots(figsize=(15, 10))
si.plot_probe_map(raw_rec, ax=ax, with_channel_ids=True)
ax.set_ylim(-100, 100)
plt.show()

# High-pass filter the data
rec1 = si.highpass_filter(raw_rec, freq_min=300.)

# Identify bad channels and remove them
bad_channel_ids, channel_labels = si.detect_bad_channels(rec1)
rec2 = rec1.remove_channels(bad_channel_ids)
print('bad_channel_ids', bad_channel_ids)

# Phase shift the data to account for neuropixel sampling offset
rec3 = si.phase_shift(rec2)

# Common median reference the data
rec4 = si.common_reference(rec3, operator="median", reference="global")
rec = rec4
print('rec',rec)

# Look at the data across multiple steps of pre-processing
si.plot_traces({'filter':rec1, 'cmr': rec4}, backend='ipywidgets')
fig, axs = plt.subplots(ncols=3, figsize=(20, 10))
si.plot_traces(rec1, backend='matplotlib',  clim=(-50, 50), ax=axs[0])
si.plot_traces(rec4, backend='matplotlib',  clim=(-50, 50), ax=axs[1])
si.plot_traces(rec, backend='matplotlib',  clim=(-50, 50), ax=axs[2])
for i, label in enumerate(('filter', 'cmr', 'final')):
    axs[i].set_title(label)

fig, ax = plt.subplots(figsize=(20, 10))
some_chans = rec.channel_ids[[100, 150, 200, ]]
si.plot_traces({'filter':rec1, 'cmr': rec4}, backend='matplotlib', mode='line', ax=ax, channel_ids=some_chans)
```

The error is:

AttributeError: module 'spikeinterface.full' has no attribute 'plot_traces'

I removed `%matplotlib inline` and `%matplotlib widget` because they were causing errors, but perhaps they are necessary?

I'd appreciate any assistance you could provide. Thank you!

~Nick

alejoe91, 2023-08-01:
Hi Nick!

As parto of a recent PR (#1859) we renamed the `plot_timeseries` function to `plot_traces` (we use traces everywhere else in the package!). The documentation is built from main, and that's why it reflects this change which is still unreleased.

So you should be good to go if you change the `plot_traces` to `plit_timeseries`

nmtimme, 2023-08-02:
That fixed it. Thank you!

closed 2023-08-02 by nmtimme