SpikeInterface/spikeinterface #1901: Mountainsort5 extraction fails as expects `.npy`

JoeZiminski, 2023-08-02:
I may be doing something wrong, trying to load sorting output created with `mountainsort5` option with:

`sorting = si.extractors.load_mda_sorting(file_path, sampling_frequency)

fails because `mdaextractors.py` expects a `.npy` file but the output file is `.npz`. It seems like loading is handled with `NpzSortingExtractor` [here](https://github.com/SpikeInterface/spikeinterface/blob/f5af76ef1a97732b2bcbec015d366ace15b72cc9/src/spikeinterface/sorters/external/mountainsort5.py#L185) - should this case be added to `mdaextractors`? Happy to open a PR if required / this is a good way forward.

alejoe91, 2023-08-21:
Hi @JoeZiminski 

Each sorter handles its own format. MS5 uses the NPZ format from SI, not MDA anymore. To make sure you use the correct format, you can do:

```
import spikeinterface.sorters as ss

sorting = ss.Mountainsort5Sorter.get_result_from_folder(output_folder)
```

JoeZiminski, 2023-08-29:
That's great, thanks @alejoe91!