SpikeInterface/spikeinterface #1906: Issue saving agreement sorting object to disk

rbedfordwork, 2023-08-03:
I want to run extract_waveforms on an AgreementSortingExtractor object. According to issue #972 I need to save the agreement sorting object to disk first. However, this returns an error saying the object is not JSON serializable.

Here's my full code:

`
mcmp = sc.compare_multiple_sorters(
    sorting_list=[klusta,mountainsort5,pykilosort],
    name_list=['klusta','mountainsort5','pykilosort'],
    verbose=True)

agr = mcmp.get_agreement_sorting(minimum_agreement_count=2)

agr = agr.save(folder='agreement_sorting')
`

And here's the full error message:

`
Traceback (most recent call last):
  File "/tmp/ipykernel_2498083/3956134603.py", line 3, in <module>
    agr2 = agr.save(folder='agreement_sorting')
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lmb/home/rbedford/.local/mambaforge/envs/spikeinterface/lib/python3.11/site-packages/spikeinterface/core/base.py", line 749, in save
    loaded_extractor = self.save_to_folder(**kwargs)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lmb/home/rbedford/.local/mambaforge/envs/spikeinterface/lib/python3.11/site-packages/spikeinterface/core/base.py", line 818, in save_to_folder
    self.dump(provenance_file)
  File "/lmb/home/rbedford/.local/mambaforge/envs/spikeinterface/lib/python3.11/site-packages/spikeinterface/core/base.py", line 558, in dump
    self.dump_to_json(file_path, relative_to=relative_to, folder_metadata=folder_metadata)
  File "/lmb/home/rbedford/.local/mambaforge/envs/spikeinterface/lib/python3.11/site-packages/spikeinterface/core/base.py", line 596, in dump_to_json
    json.dumps(dump_dict, indent=4, cls=SIJsonEncoder),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lmb/home/rbedford/.local/mambaforge/envs/spikeinterface/lib/python3.11/json/__init__.py", line 238, in dumps
    **kw).encode(obj)
          ^^^^^^^^^^^
  File "/lmb/home/rbedford/.local/mambaforge/envs/spikeinterface/lib/python3.11/json/encoder.py", line 202, in encode
    chunks = list(chunks)
             ^^^^^^^^^^^^
  File "/lmb/home/rbedford/.local/mambaforge/envs/spikeinterface/lib/python3.11/json/encoder.py", line 432, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/lmb/home/rbedford/.local/mambaforge/envs/spikeinterface/lib/python3.11/json/encoder.py", line 406, in _iterencode_dict
    yield from chunks
  File "/lmb/home/rbedford/.local/mambaforge/envs/spikeinterface/lib/python3.11/json/encoder.py", line 406, in _iterencode_dict
    yield from chunks
  File "/lmb/home/rbedford/.local/mambaforge/envs/spikeinterface/lib/python3.11/json/encoder.py", line 439, in _iterencode
    o = _default(o)
        ^^^^^^^^^^^
  File "/lmb/home/rbedford/.local/mambaforge/envs/spikeinterface/lib/python3.11/site-packages/spikeinterface/core/core_tools.py", line 106, in default
    return super().default(obj)
           ^^^^^^^^^^^^^^^^^^^^
  File "/lmb/home/rbedford/.local/mambaforge/envs/spikeinterface/lib/python3.11/json/encoder.py", line 180, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
TypeError: Object of type MultiSortingComparison is not JSON serializable
`

I am using SpikeInterface 0.98.2.

Any help with this would be much appreciated! Thanks in advance.

alejoe91, 2023-08-03:
Thanks for the report! We'll take a look at this. 

What's the output of : 
```
agr.check_if_json_serializable()
```
?

rbedfordwork, 2023-08-04:
No the output is True.

alejoe91, 2023-08-04:
Ok, that's the issue! We'll fix it

rbedfordwork, 2023-08-04:
That's great thank you!

rbedfordwork, 2023-08-04:
For anyone else who encounters this issue you can use a very hacky fix by just setting agr._is_json_serializable = False, and you can then run agr.save() then si.extract_waveforms() just fine. I'll use this until the bug is fixed!

alejoe91, 2023-08-04:
Strange because it should be false! https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/comparison/multicomparisons.py#L231

rbedfordwork, 2023-08-04:
I think the issue might be that BaseSorting.__init__() is run after self._is_json_serializable = False, which in turn calls BaseExtractor__init__() which overrides to self._is_json_serializable = True.

closed 2023-08-22 by h-mayorquin