SpikeInterface/spikeinterface #1910: Kilosort3 Hangs

nmtimme, 2023-08-04:
Hello

I'm trying to go through the guide to using spike interface to spike sort neuropixel data (https://spikeinterface.readthedocs.io/en/latest/how_to/analyse_neuropixels.html), but I'm running into an error with detect_peaks. Thank you for quickly addressing my issue with plot_traces #1899 ! I'm still working on the suggestions for the radius_um error #1907 .

I'm running version 0.98.2 on a Linux cluster with a GPU. I'm running on a virtual desktop, so I can view plots and interact with GUIs. I installed spikeinterface using miniconda 4.12.0 using this .yml file:

```
name: si_env
channels:
  - conda-forge
  - defaults
dependencies:
  - python=3.9
  - pip>=21.0
  - mamba
  # numpy 1.22 break numba which break tridesclous
  - numpy<1.22
  # joblib 1.2 is breaking hdbscan
  - joblib=1.1
  - tqdm
  - matplotlib
  - h5py
  - pandas
  - scikit-learn
  - networkx
  - pybind11
  - loky
  - hdbscan
  - numba
  - jupyter
  - mpi4py
  - cxx-compiler
  - libxcb
  - pyqt=5
  - pyqtgraph
  - htop
  - ipywidgets
  - ipympl
  - pip:
    - ephyviewer
    - neo>=0.11
    - elephant>=0.10.0
    - probeinterface>=0.2.11
    - MEArec>=1.8
    - spikeinterface[full, widgets]
    - spikeinterface-gui
    - tridesclous>=1.6.6.1
    - spyking-circus>=1.1.0
    # - phy==2.0b5
```

After creating the environment, I also installed spython and cuda-python to this environment based on the page on containerized sorters (https://spikeinterface.readthedocs.io/en/latest/modules/sorters.html#containerizedsorters) and some issues I looked through. I have cudatoolkit 11.7 and singularity 3.8.3 installed also. 

I am running the following script:

```
import spikeinterface.full as si
import spikeinterface.widgets as siw
print(si.__version__)

import numpy as np
import matplotlib.pyplot as plt
from pathlib import Path

# Define the path to the data
base_folder = Path('/NickPilotNeuropixelEphysData')
openephys_folder = base_folder / '2023-03-15_15-18-59'

# Figure out the data streams in this data set
stream_names, stream_ids = si.get_neo_streams('openephys', openephys_folder)
[... 128 lines cut ...]
  File "/.conda/envs/si_env/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py", line 329, in run_command
    res = Client.execute(self.client_instance, command, options=options)
  File "/.conda/envs/si_env/lib/python3.9/site-packages/spython/main/execute.py", line 110, in execute
    return self._run_command(
  File "/.conda/envs/si_env/lib/python3.9/site-packages/spython/main/base/command.py", line 137, in run_command
    result = run_cmd(
  File "/.conda/envs/si_env/lib/python3.9/site-packages/spython/utils/terminal.py", line 196, in run_command
    for line in process.communicate():
  File "/.conda/envs/si_env/lib/python3.9/subprocess.py", line 1134, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
  File "/.conda/envs/si_env/lib/python3.9/subprocess.py", line 1979, in _communicate
    ready = selector.select(timeout)
  File "/.conda/envs/si_env/lib/python3.9/selectors.py", line 416, in select
    fd_event_list = self._selector.poll(timeout)
```

Given the behavior and the name of the timeout variable, I thought this might be relevant to understanding what is happening.

The machine has an nvidia A100 GPU and I think I've done what I should to make sure it is being utilized (e.g., cudatoolkit installed), but I don't know if it is being utilized. 

I don't see kilosort3 on the list of installed sorters, so perhaps I missed some step in terms of properly installing the containerized version of kilosort3?

I'd appreciate any ideas on how to figure out what is happening, address this problem, and/or determine if the GPU is being utilized. Thank you!

~Nick

zm711, 2023-08-04:
[image]
@nmtimme, I'm not super familiar with the containerized running of SI, but based on Kilosort's own metrics (https://www.biorxiv.org/content/10.1101/2023.01.07.523036v1.full.pdf+html) the time to run Kilosort3 can be quite substantial. Could you test the same file using Kilosort2 or 2.5 and see if those hang or not? Or could you book a longer appointment on your cluster? And if that doesn't work we can wait for the experts to troubleshoot further.

As far as I know Kilosort_1,2,2.5,3 will only run with GPU (ie they will completely fail if GPU is not being accessed) so if you are hanging and not explicitly erroring it seems like things should be engaging.

nmtimme, 2023-08-05:
Hi @zm711 , thanks for that suggestion. Unfortunately, I tried it with Kilosort 2.5 and I got the same result. I let it run for about 2 hours. It hangs up at:

`Running kilosort2_5 sorter inside spikeinterface/kilosort2_5-compiled-base:latest`

No files were produced, other than those two json files, the site mapping matlab file, and the .dat file that are produced with a few minutes of starting the sorting, just like with kilosort3. Then, when I forced it to stop, I got the exact same Traceback as in my original post. I'd appreciate any other suggestions people might have. Thanks!

~Nick

nmtimme, 2023-08-10:
I believe I have resolved the problem. I found that if I requested 128 GB of RAM on the cluster I am using, kilosort3 ran just fine in about 25 minutes. I'm not sure why failing to request a specific amount caused kilosort3 to hang up or slow way down rather than produce some sort of error, but that seems to have been the result. Thanks!

closed 2023-08-10 by nmtimme

zm711, 2023-08-10:
@nmtimme Super interesting and good to know. Now that you mention it, in the past we were running Kilosort2 locally but with data from a storage server and once when there were server upgrades our transfer speeds tanked and Kilosort2 (which normally was taking 20-30 minutes for us) ended up plodding along for 4 or 5 hours. It finally finished the run so I don't think it is actually erroring; it just works at the speed allowable by the current computer config.