SpikeInterface/spikeinterface #1912: Overflow with spre.whiten

ysueoka, 2023-08-07:
I am using the code snippet below to preprocess the neural data for mountainsort5. 

    recording = se.read_mda_recording(mda_path, raw_fname)    
    recording_filtered = spre.bandpass_filter(recording, freq_min=freq_min, freq_max=freq_max)
    recording_preprocessed: si.BaseRecording = spre.whiten(recording_filtered, dtype='float32')
    sorting = ms5.sorting_scheme2(recording_preprocessed,
            sorting_parameters=ms5.Scheme2SortingParameters( ...
            )
        )

The code fails on the third line, saying "NoneType object is not subscriptable (mdaextractors.py line 156)." This seems to be due to an overflow error in _read_chunk_1d (line390 in mdaextractors.py) when the input i is above 10^8, which happens in the first~fourth time  (changes every time) the _read_chunk_1d is called. Does this mean the neural data is too big? If so is there a good way to chunk the data, run the whitening, and reassemble it into one file? For context, the neural data is stored in mda format as mda_path/raw_fname, with a file size of 2.7GB.

ysueoka, 2023-08-09:
I was able to fix this by changing the get_random_data_chunks in si.core.recording_tools. Since the start_frame (line 46) was in type int32, it threw an overflow error when the frame number was too big. Adding a line below casted the start_frame to int type, and fixed the issue

   start_frame = start_frame.item()

samuelgarcia, 2023-08-28:
Hi thanks for the catch.
Can you make a PR for this ?

ysueoka, 2023-08-29:
Seems like the issue has been addressed by others, and the code in the main branch no longer has this issue!

zm711, 2023-10-06:
@samuelgarcia
looks like this particular error was fixed. Do you want to close or leave open to look for other cases of int32 which may overflow?

closed 2023-10-06 by samuelgarcia