SpikeInterface/spikeinterface #1913: How to make json file in reading mda files

YinJiapeng, 2023-08-08:
I am using **_**_spikeinterface.extractors.read_mda_recording_**_** to read *.mda files.

[image]

Now I have 2 questions:
1. I don't know how to get the params.json file. How to produce a params.json file for mda file processing?
​
2. Is there any example geom.csv file? So I can produce one geom file following the example.

Many thanks.

alejoe91, 2023-08-21:
Hi @YinJiapeng 

Where did you get the `.mda` files? The `params.json` should be part of the folder. 

The only piece of information needed on the SpikeInterface side is the sampling frequency.

Here is a sample `params.json` that should work:

```
{"samplerate": 30000}
```

The `geom.csv` is is a comma separated text file with x, y position for each electrode, e.g.:
```
0, 0
0, 100
0, 200
0, 300
...
```

li1901, 2023-09-14:
> Hi  你好@YinJiapeng
> 
> Where did you get the `.mda` files? The `params.json` should be part of the folder.您从哪里获取 `.mda` 文件？ `params.json` 应该是文件夹的一部分。
> 
> The only piece of information needed on the SpikeInterface side is the sampling frequency.SpikeInterface 端所需的唯一信息是采样频率。
> 
> Here is a sample `params.json` that should work:这是一个应该可以工作的示例 `params.json` ：
> 
> ```
> {"samplerate": 30000}
> ```
> 
> The `geom.csv` is is a comma separated text file with x, y position for each electrode, e.g.: `geom.csv` 是一个逗号分隔的文本文件，每个电极都有 x、y 位置，例如：
> 
> ```
> 0, 0
> 0, 100
> 0, 200
> 0, 300
> ...
> ```
That's awesome! But I have a question about more specific details. Electrodes typically have physical dimensions. How do you define the XY position between them? Do you use the center point of their geometric dimensions?What should be the distance unit in the document? Meters or micrometers?thanku so much

alejoe91, 2023-09-14:
Generally, you would use the center of the electrode and express dimensions as um. Note that we use a separate package, [probeinterface](https://probeinterface.readthedocs.io/en/main/), to handle the probe. It has a much richer representation of the probe, so feel free to take a look and use it in your processing

closed 2023-11-06 by alejoe91