SpikeInterface/spikeinterface #1915: Error using run_sorter

Sietse-k, 2023-08-09:
Hi, I'm working on the "get started" spikeInterface guide (https://spikeinterface.readthedocs.io/en/latest/how_to/get_started.html). Using simulated data. However, I run into the following problem:

sorting_TDC = ss.run_sorter(sorter_name="tridesclous", recording=recording_preprocessed, detect_threshold=4)
print(sorting_TDC)

Traceback (most recent call last):
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\IPython\core\interactiveshell.py", line 3508, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-3-777f8a455281>", line 1, in <module>
    sorting_TDC = ss.run_sorter(sorter_name="tridesclous", recording=recording_preprocessed, detect_threshold=4)
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py", line 148, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py", line 174, in run_sorter_local
    SorterClass.run_from_folder(output_folder, raise_error, verbose)
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\basesorter.py", line 269, in run_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\basesorter.py", line 234, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\external\tridesclous.py", line 149, in _run_from_folder
    tdc.apply_all_catalogue_steps(cc, catalogue_nested_params, verbose=verbose)
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\tridesclous\cataloguetools.py", line 119, in apply_all_catalogue_steps
    cc.find_clusters(method=params['cluster_method'], recompute_centroid=False, order=False, **params['cluster_kargs'])
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\tridesclous\catalogueconstructor.py", line 1173, in find_clusters
    labels = cluster.find_clusters(self, method=method, selection=selection, **kargs)
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\tridesclous\cluster.py", line 164, in find_clusters
    labels = pruningshears.do_the_job()
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\tridesclous\pruningshears.py", line 135, in do_the_job
    cluster_labels = self.explore_split_loop()
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\tridesclous\pruningshears.py", line 511, in explore_split_loop
    labels_l0 = self.one_sub_cluster(features_l0, allow_single_cluster=False)
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\tridesclous\pruningshears.py", line 237, in one_sub_cluster
    local_labels = clusterer.fit_predict(local_features)
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\hdbscan\hdbscan_.py", line 1243, in fit_predict
    self.fit(X)
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\hdbscan\hdbscan_.py", line 1205, in fit
    ) = hdbscan(clean_data, **kwargs)
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\hdbscan\hdbscan_.py", line 884, in hdbscan
    _tree_to_labels(
  File "C:\Users\Sietse\anaconda3\envs\si_env\lib\site-packages\hdbscan\hdbscan_.py", line 80, in _tree_to_labels
    labels, probabilities, stabilities = get_clusters(
  File "hdbscan\\_hdbscan_tree.pyx", line 659, in hdbscan._hdbscan_tree.get_clusters
  File "hdbscan\\_hdbscan_tree.pyx", line 733, in hdbscan._hdbscan_tree.get_clusters
TypeError: 'numpy.float64' object cannot be interpreted as an integer
Spike sorting failed. You can inspect the runtime trace in C:\Users\Sietse\PycharmProjects\spikesorting\tridesclous_output/spikeinterface_log.json.

Any help would be greatly appreciated.

zm711, 2023-08-10:
I was able to recreate this using the toy dataset with an up-to-date environment 

```bash
python 3.10, 
tridesclous 1.6.8,
hdbscan 0.8.33, 
si 0.98.2
sklearn 1.3.0
numpy 1.24.
```

Error for toy dataset
```
TypeError: 'numpy.float64' object cannot be interpreted as an integer
```

If I down grade numpy, hdbscan and sklearn then tridesclous worked fine.

```
numpy 1.23.5
scikit-learn 1.1.3
hdbscan 0.8.29
```

I was also trying this on a Windows computer, so maybe there is a Windows specific issue with the si install. I'll keep digging.

Sietse-k, 2023-08-11:
Hi Zach, thanks for your reply. I'll try those specific versions in that case. I also tried running trideclous inside docker, and that did work for me. Let me know if you find the specific issue. 
Cheers.

alejoe91, 2023-08-11:
Thanks for debugging this. This should be fixed on the hdbscan side, hopefully soon!

zm711, 2023-08-11:
Based on my reading `8.33` was suppose to be the fix, but it sounds like there are some `cython` versioning issues with hdbscan, so I agree hopefully they fix it soon! In the issues everyone just does varying reinstalls with different mixes of `python`, `numpy`, `hdbscan` and `sklearn` and different combos work for different people.

alejoe91, 2023-08-21:
Can you try to make sure you have the latest Cython version installed? `pip install -U Cython`

zm711, 2023-08-21:
@alejoe91, I just checked the issue again and the maintainers just weighed in

> This is really quirky, and I am having a great deal of trouble reproducing it in a way that I can actually debug it myself.

and their advice:

>Did you try a fresh environment?

> ```python
>conda create -n testenv python=3.11
>pip install hdbscan==0.8.33 numpy==1.24.4 notebook==7.0.2 scikit-learn==1.3.0
>```

>Cython should be something like 0.29.26 not 3.0.

>If there's a hdbscan error, try:

>```python 
> pip install --upgrade git+https://github.com/scikit-learn-contrib/hdbscan.git#egg=hdbscan
>```

zm711, 2023-08-25:
hdbscan released a conda patch yesterday that they hope fixed it. The `hdbscan` should be version `0.8.33_3` (I copied from their issue so the `_3` is what they have written).  You can either reinstall the overall environment or try to just update the hdbscan itself up to the latest version.

zm711, 2023-08-29:
@alejoe91 @samuelgarcia 

This is fixed now.  Users should see this tagline to know that they have the correct hdbscan installed
```
hdbscan                   0.8.33          py310h3e78b6c_3    conda-forge
```

The `_3` is attached to the build number which corrects for the incorrect cython during the build in the previous versions.

I was able to fully run tridesclous with the toy data set without error. 

Just for completeness
```
numpy                     1.24.4          py310hd02465a_0    conda-forge
numba                     0.57.1          py310h19bcfe9_0    conda-forge
scikit-learn              1.3.0           py310hd266714_0    conda-forge
scipy                     1.11.2          py310h578b7cb_0    conda-forge
spikeinterface            0.98.2                   pypi_0    pypi
tridesclous               1.6.8                    pypi_0    pypi
```

samuelgarcia, 2023-08-29:
Cool thanks for testing.

zm711, 2023-10-06:
@Sietse-k , Just wanted to make sure you were able to test this and get this working with the hdbscan. Are you having any other issues with install or is this fixed for you now?

zm711, 2023-11-06:
@Sietse-k, On our end this is all working since they have updated hdbscan. If you run into new issues installing sorters or with the package feel free to open a new issue.

closed 2023-11-06 by zm711