SpikeInterface/spikeinterface #1916: Thoughts on extracting waveforms from Kilosort's `temp_wh.dat`

JoeZiminski, 2023-08-10:
Currently (I believe) the canonical way (e..g in the docs) to pipeline sorting is 

1) generate a `recording` object for preprocessing
2) pass this to a sorter and get a sorting object
3) extract waveforms using the preprocessing recording and sorting output. The waveform data is extracted from the preprocessed recording.

(please let me know if this is wrong off the bat)

One problem with this workflow is that in some instances the sorter performs it's own preprocessing, in the case of Kilosort this is saved as the `temp_wh.dat` file. In this case, the waveforms saved by spikeinterface will not be the same as those in Phy / the true data that was sorted on. I guess this is similar in the case that any sorter does it's own preprocessing, although I am not sure of the specifics of other sorters.

I wonder what the best approach for this is (if it is indeed a problem) - could it be solved by documentation? Or, should the sorting object hold a reference to the output of a sorter's preprocessing steps (e.g. `temp_wh.dat` loaded as an SI recording object in the Kilosort case) which waveform extraction can use under the good? For example instead of:

```
recording = load_data(...)
prepro_recording = SomePreproSteps(recording)
sorting = run_sorter(prepro_recording)
we = extract_waveforms(prepro_recording, sorting, ...)
```
the call to `extract_waveforms1` would be:
```
we = extract_waveforms(sorting, ...)
```
and under the hood sorting would use `temp_wh.dat` to extract waveforms from, or whatever other sorter-specific preprocesing file there is. Otherwise, it could load the preprocessing file directly from the metadata in the sorting output, or optionally take a recording object as is done now.

claudio1212, 2023-08-15:
I'm having a similar problem, in that I apparently cannot run sorting with kilosort using the skip kilosort preprocessing option (it gives a "negative dimensions are not allowed" error), so I'd have to run it with raw recording data and then re do preprocessing to extract waveforms, which appears to me to be somewhat wrong. So I agree with your request and would like to see if there are some solutions

alejoe91, 2023-08-21:
Hi @JoeZiminski 

You can already load the KS pre-processed data as a plain binary file with the `si.read_binary()` function:

The main "problem" is that the preprocessed data will not contain all channels, as some are removed for lack of activity. 
I guess that we could make a `KilosortTempWhRecording` that loads the preprocessed data and reads the metadata (such as which channels have been removed) from the KS output. Then one can extract waveforms with whatever preferred recording. What do you think?

JoeZiminski, 2024-05-13:
Closed further to discussion on #1986 and #1954  as it is not very feasible.

closed 2024-05-13 by JoeZiminski