SpikeInterface/spikeinterface #1924: Errors running installed sorters

rdzakpasu, 2023-08-17:
I am working through https://spikeinterface.readthedocs.io/en/latest/how_to/get_started.html. When I get to the point where I try to run tridesclous, I get the following errors:

SpikeSortingError                         Traceback (most recent call last)
Cell In[30], line 1
----> 1 sorting_TDC = ss.run_sorter(sorter_name="tridesclous", recording=recording_preprocessed, detect_threshold=4)
      2 print(sorting_TDC)

File ~\AppData\Local\anaconda3\Lib\site-packages\spikeinterface\sorters\runsorter.py:148, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, **sorter_params)
    141             container_image = singularity_image
    142     return run_sorter_container(
    143         container_image=container_image,
    144         mode=mode,
    145         **common_kwargs,
    146     )
--> 148 return run_sorter_local(**common_kwargs)

File ~\AppData\Local\anaconda3\Lib\site-packages\spikeinterface\sorters\runsorter.py:174, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    172 SorterClass.set_params_to_folder(recording, output_folder, sorter_params, verbose)
    173 SorterClass.setup_recording(recording, output_folder, verbose=verbose)
--> 174 SorterClass.run_from_folder(output_folder, raise_error, verbose)
    175 if with_output:
    176     sorting = SorterClass.get_result_from_folder(output_folder)

File ~\AppData\Local\anaconda3\Lib\site-packages\spikeinterface\sorters\basesorter.py:269, in BaseSorter.run_from_folder(cls, output_folder, raise_error, verbose)
    266         print(f"{sorter_name} run time {run_time:0.2f}s")
    268 if has_error and raise_error:
--> 269     raise SpikeSortingError(
    270         f"Spike sorting error trace:\n{log['error_trace']}\n"
    271         f"Spike sorting failed. You can inspect the runtime trace in {output_folder}/spikeinterface_log.json."
    272     )
    274 return run_time

SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "C:\Users\rd259\AppData\Local\anaconda3\Lib\site-packages\spikeinterface\sorters\basesorter.py", line 234, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "C:\Users\rd259\AppData\Local\anaconda3\Lib\site-packages\spikeinterface\sorters\external\tridesclous.py", line 149, in _run_from_folder
    tdc.apply_all_catalogue_steps(cc, catalogue_nested_params, verbose=verbose)
  File "C:\Users\rd259\AppData\Local\anaconda3\Lib\site-packages\tridesclous\cataloguetools.py", line 119, in apply_all_catalogue_steps
    cc.find_clusters(method=params['cluster_method'], recompute_centroid=False, order=False, **params['cluster_kargs'])
  File "C:\Users\rd259\AppData\Local\anaconda3\Lib\site-packages\tridesclous\catalogueconstructor.py", line 1173, in find_clusters
    labels = cluster.find_clusters(self, method=method, selection=selection, **kargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\rd259\AppData\Local\anaconda3\Lib\site-packages\tridesclous\cluster.py", line 164, in find_clusters
    labels = pruningshears.do_the_job()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\rd259\AppData\Local\anaconda3\Lib\site-packages\tridesclous\pruningshears.py", line 135, in do_the_job
    cluster_labels = self.explore_split_loop()
                     ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\rd259\AppData\Local\anaconda3\Lib\site-packages\tridesclous\pruningshears.py", line 511, in explore_split_loop
    labels_l0 = self.one_sub_cluster(features_l0, allow_single_cluster=False)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\rd259\AppData\Local\anaconda3\Lib\site-packages\tridesclous\pruningshears.py", line 237, in one_sub_cluster
    local_labels = clusterer.fit_predict(local_features)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\rd259\AppData\Local\anaconda3\Lib\site-packages\hdbscan\hdbscan_.py", line 1243, in fit_predict
    self.fit(X)
  File "C:\Users\rd259\AppData\Local\anaconda3\Lib\site-packages\hdbscan\hdbscan_.py", line 1205, in fit
    ) = hdbscan(clean_data, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\rd259\AppData\Local\anaconda3\Lib\site-packages\hdbscan\hdbscan_.py", line 884, in hdbscan
    _tree_to_labels(
  File "C:\Users\rd259\AppData\Local\anaconda3\Lib\site-packages\hdbscan\hdbscan_.py", line 80, in _tree_to_labels
    labels, probabilities, stabilities = get_clusters(
                                         ^^^^^^^^^^^^^
  File "hdbscan\\_hdbscan_tree.pyx", line 659, in hdbscan._hdbscan_tree.get_clusters
  File "hdbscan\\_hdbscan_tree.pyx", line 733, in hdbscan._hdbscan_tree.get_clusters
TypeError: 'numpy.float64' object cannot be interpreted as an integer

Spike sorting failed. You can inspect the runtime trace in C:\Users\rd259\tridesclous_output/spikeinterface_log.json.

The same thing happens when I try to run spykingcircus2.  Thank you for your help.

zm711, 2023-08-17:
@rdzakpasu 
That is an issue with hdbscan. They need to implement a fix on their side (hopefully soon). #1915 goes through some testing that might help. Basically you need to play around with the version number of hdbscan, numpy, scikit-learn, (and maybe python). I got
```
numpy 1.23.5
scikit-learn 1.1.3
hdbscan 0.8.29
``` 
to work with python 3.10, but depending on your machine you may need to play around a bit.

rdzakpasu, 2023-08-18:
Thank you, Zach, for your reply. I will try this and look at #1915 to read more.

zm711, 2023-08-29:
@rdzakpasu 
Just as an FYI hdbscan has fixed the issue. You need to reinstall from conda-forge. It should say:

```
Name                  Version         Build              Channel
hdbscan               0.8.33          py310h3e78b6c_3    conda-forge
```

if you type `conda list` in your spikeinterface environment. The key change is the build needs the `_3`.

rdzakpasu, 2023-08-30:
Good news! I will try it out. Thanks for letting me know.

zm711, 2023-10-06:
@rdzakpasu 

Were you able to test to make sure it is now working for you? Just want to make sure the new build of hdbscan works on a "real" dataset.

rdzakpasu, 2023-10-10:
Thank you for checking back with us! Unfortunately, the semester has begun
and we're swamped with teaching. It's on my todo list though; once I test
it I can let you know.

Best,

Rhonda

alejoe91, 2023-11-06:
@rdzakpasu closing for now. Feel free to re-open if that's still an issue when you get the chance to test this out

closed 2023-11-06 by alejoe91