SpikeInterface/spikeinterface #1925: Problems running kilosort3 on individual shanks of a custom probe

nmtimme, 2023-08-18:
Hello

I am trying to sort open ephys data gathered from an F-series 64 channel Cambridge probe. Unfortunately, this probe and intan headstage combination are not already available in probe interface, so we made our own probe. However, I suspect we are not doing this properly because we are receiving errors if we run the containerized version of kilosort3 on the data either as a whole or as individual shanks. I'm fairly confident kilosort3 is working properly in the containerized version because we are able to successfully spike sort neuropixel data.

This is the python script we are running (spike interface version 0.98.2):

```
import spikeinterface.full as si
import probeinterface as pi
import spikeinterface.widgets as siw
print(si.__version__)

import numpy as np
import matplotlib.pyplot as plt
from pathlib import Path
from probeCreator import probeCreator

# Define the path to the data
base_folder = Path('/N/project/lapishLabWorkspace/Nick2CAPTestData')
openephys_folder = base_folder / '2018-12-11-Session4'

# Read the ephys data (stream names are 'Signals AUX', 'Signals CH', 'Signals ADC')
raw_rec = si.read_openephys(openephys_folder,stream_name='Signals CH')
print('raw_rec',raw_rec)

# Create the probe
probe = probeCreator('NT2CAPCohort2Animal5')
# probe = probeCreator('TestAnimal')
print('Probe:',probe)
# print('Contact Positions:',probe.contact_positions)

# Set the probe into the recording
raw_rec.set_probe(probe, in_place=True, group_mode="by_shank")
# raw_rec.set_probe(probe)
# raw_rec.set_channel_locations(probe.contact_positions)
# print('rec Contact Positions:',raw_rec.get_channel_locations())

# High-pass filter the data
rec1 = si.highpass_filter(raw_rec, freq_min=300.)

# Identify bad channels and remove them
bad_channel_ids, channel_labels = si.detect_bad_channels(rec1)
rec2 = rec1.remove_channels(bad_channel_ids)
print('bad_channel_ids', bad_channel_ids)

# Common median reference the data
rec3 = si.common_reference(rec2, operator="median", reference="global")
rec = rec3
print('rec',rec)

# Look at the rasters at various stages of preprocessing
fig, axs = plt.subplots(ncols=3, figsize=(20, 10))
si.plot_timeseries(rec1, backend='matplotlib',  clim=(-50, 50), ax=axs[0])
si.plot_timeseries(rec3, backend='matplotlib',  clim=(-50, 50), ax=axs[1])
si.plot_timeseries(rec, backend='matplotlib',  clim=(-50, 50), ax=axs[2])
for i, label in enumerate(('filter', 'cmr', 'final')):
    axs[i].set_title(label)
# plt.show() # Uncomment to show plot

# Look at some example traces
fig, ax = plt.subplots(figsize=(20, 10))
some_chans = rec.channel_ids[[5, 10, 20, ]]
si.plot_timeseries({'filter':rec1, 'cmr': rec3}, backend='matplotlib', mode='line', ax=ax, channel_ids=some_chans)
# plt.show() # Uncomment to show plot

# Estimate the noise on the scaled traces (microV) or on the raw one (which is in our case int16).
noise_levels_microV = si.get_noise_levels(rec, return_scaled=True)
noise_levels_int16 = si.get_noise_levels(rec, return_scaled=False)
fig, ax = plt.subplots()
_ = ax.hist(noise_levels_microV, bins=np.arange(5, 30, 2.5))
[... 285 lines cut ...]
7.89 sec, 1001 batches, 1946541 spikes 
8.91 sec, 1101 batches, 2122100 spikes 
9.68 sec, 1201 batches, 2292413 spikes 
10.44 sec, 1301 batches, 2453195 spikes 
11.20 sec, 1401 batches, 2613340 spikes 
11.96 sec, 1501 batches, 2758369 spikes 
12.72 sec, 1601 batches, 2921438 spikes 
13.52 sec, 1701 batches, 3115457 spikes 
14.06 sec, 1771 batches, 3249222 spikes 
time 0.00, GROUP 1/18, units 0 
time 45.84, GROUP 6/18, units 41 
time 87.93, GROUP 11/18, units 81 
time 124.93, GROUP 16/18, units 115 
Elapsed time is 145.518486 seconds.
----------------------------------------Error using gpuArray/eig
EIG did not converge at index = 2297.
```

Based on looking at some kilosort issues, it looks like the first type of error is commonly associated with too few channels, but we find this happens on all shanks, even ones with 10 good channels. So, we're wondering if somehow kilosort isn't getting the number of channels it expects in the sort by shank method. 

The second error seems less common, but it might be related to nan or inf values popping up along the way, which we thought might be due to not giving kilosort what it expects.

We'd appreciate any suggestions people might have. Perhaps we aren't building our probe properly or connecting it with the recording? Thank you!

~Nick

nmtimme, 2023-08-22:
Hi @alejoe91 !

Thank you for those suggestions!

Regarding the proposed solution to the EIG did not converge error, perhaps this work around could be implemented in spike interface by adjusting the NT parameter in kilosort3? I see that the default value for NT is none. I'm a bit confused by this since I would think it would have a numeric value. Is the NT parameter for kilosort3 the same thing ops.NT? In any case, I tried changing NT from none to 64000 and it seems to have worked! I can sort all the channels together and not receive the EIG did not converge error. (Though, see comment below about modifying a Cambridge probe file.)

Regarding the channel map, yes, I believe I was able to verify that it is correct. However, I'm worried that I'm missing some setting along the way. For instance, I'm rather confused about what the in_place=True option is doing in set_probe. What does this do? I couldn't find much about this in the documentation, but it's possible I just missed it. 

To try to address these problems, I loaded the Cambridge probe file for the probe I'm using, I set the device_channel_indices to 0:63, and changed the positions so they corresponded to open ephys channels. I'm rather confused by the device wiring options, so I tried to work around them in this way.

All that said, I was able to reproduce the first error with a small test data set with only 9 channels generated using this code:

```
import spikeinterface.full as si
from spikeinterface.extractors import toy_example

test_recording, _ = toy_example(duration=120, seed=0, num_channels=6,num_segments=1)
test_recording = test_recording.save(folder="/N/project/lapishLabWorkspace/NickSpikeInterfaceTest")

defaultKS3Params = si.get_default_sorter_params('kilosort3')

# sorting = si.run_sorter('kilosort3', recording=test_recording, output_folder="kilosort3", singularity_image=True)
sorting = si.run_sorter('kilosort3', recording=test_recording, output_folder='/N/project/lapishLabWorkspace/NickSpikeInterfaceTest/kilosort3_output', singularity_image="spikeinterface/kilosort3-compiled-base:latest", verbose=True, **defaultKS3Params)
print(sorting)
```

Kilosort is able to run with 10 channels in this example, but I know it failed with some shanks with 10 channels in my data set, so I'm not sure if there is a hard channel number limit. I thought kilosort could run with as few as four channels, but I must have been wrong. 

~Nick

zm711, 2023-08-29:
@nmtimme 

Just wanted to drop this screenshot from the Kilosort team's latest biorxiv [paper](https://www.biorxiv.org/content/10.1101/2023.01.07.523036v1)
[image]

Although they don't give a hard cutoff they say that if the geometry (ie channel count or pitch) aren't what Kilosort likes then Kilosort 3 will just fail and for your purposes it may be better to use a different sorter (either version or completely different sorter).

nmtimme, 2023-08-29:
Thank you @zm711 ! That is good information to have about the channel count in kilosort! 

I'm still a bit curious about the NT parameter in spike interface with kilosort3 and whether this is the same as ops.NT in the original kilosort code. Can anyone clarify that? Thank you!

~Nick

zm711, 2023-08-29:
@nmtimme 

Yes it is. It is loaded into the Matlab `ops` here:

https://github.com/SpikeInterface/spikeinterface/blob/3817ee01c6dfb5c43efe16dec1dad160019874b1/src/spikeinterface/sorters/external/kilosort3.py#L206-L208

After being calculated here:

https://github.com/SpikeInterface/spikeinterface/blob/3817ee01c6dfb5c43efe16dec1dad160019874b1/src/spikeinterface/sorters/external/kilosort3.py#L135-L140

set with a default of `None` to allow `SpikeInterface` to calculate the value itself.

You can see the default params `SpikeInterface` uses with the function `get_default_sorter_params` and to see the descriptions of these parameters you can run `get_sorter_params_descriptions`. These can be useful functions to see what params you as the user can change (although it is not always necessarily the best to change them :))
For example:

```bash
>>> si.get_default_sorter_params('kilosort3')
{'detect_threshold': 6, 'projection_threshold': [9, 9], 'preclust_threshold': 8, 'car': True, 'minFR': 0.2, 'minfr_goodchannels': 0.2, 'nblocks': 5, 'sig': 20, 'freq_min': 300, 'sigmaMask': 30, 'nPCs': 3, 'ntbuff': 64, 'nfilt_factor': 4, 'do_correction': True, 'NT': None, 'AUCsplit': 0.8, 'wave_length': 61, 'keep_good_only': False, 'skip_kilosort_preprocessing': False, 'scaleproc': None, 'save_rez_to_mat': False, 'delete_tmp_files': True, 'delete_recording_dat': False, 'n_jobs': 8, 'chunk_duration': '1s', 'progress_bar': True}
>>> si.get_sorter_params_descriptions('kilosort3')
{'detect_threshold': 'Threshold for spike detection', 'projection_threshold': 'Threshold on projections', 'preclust_threshold': 'Threshold crossings for pre-clustering (in PCA projection space)', 'car': 'Enable or disable common reference', 'minFR': 'Minimum spike rate (Hz), if a cluster falls below this for too long it gets removed', 'minfr_goodchannels': "Minimum firing rate on a 'good' channel", 'nblocks': "blocks for registration. 0 turns it off, 1 does rigid registration. Replaces 'datashift' option.", 'sig': 'spatial smoothness constant for registration', 'freq_min': 'High-pass filter cutoff frequency', 'sigmaMask': 'Spatial constant in um for computing residual variance of spike', 'nPCs': 'Number of PCA dimensions', 'ntbuff': 'Samples of symmetrical buffer for whitening and spike detection', 'nfilt_factor': 'Max number of clusters per good channel (even temporary ones) 4', 'do_correction': 'If True drift registration is applied', 'NT': 'Batch size (if None it is automatically computed)', 'AUCsplit': 'Threshold on the area under the curve (AUC) criterion for performing a split in the final step', 'wave_length': 'size of the waveform extracted around each detected peak, (Default 61, maximum 81)', 'keep_good_only': "If True only 'good' units are returned", 'skip_kilosort_preprocessing': 'Can optionaly skip the internal kilosort preprocessing', 'scaleproc': 'int16 scaling of whitened data, if None set to 200.', 'save_rez_to_mat': 'Save the full rez internal struc to mat file', 'delete_tmp_files': 'Whether to delete all temporary files after a successful run', 'delete_recording_dat': "Whether to delete the 'recording.dat' file after a successful run", 'n_jobs': 'Number of jobs (when saving ti binary) - default -1 (all cores)', 'chunk_size': 'Number of samples per chunk (when saving ti binary) - default global', 'chunk_memory': "Memory usage for each job (e.g. '100M', '1G') (when saving to binary) - default global", 'total_memory': "Total memory usage (e.g. '500M', '2G') (when saving to binary) - default global", 'chunk_duration': "Chunk duration in s if float or with units if str (e.g. '1s', '500ms') (when saving to binary) - default global", 'progress_bar': 'If True, progress bar is shown (when saving to binary) - default global'}
```

Zach

nmtimme, 2023-08-29:
@zm711 

That is very helpful! Thank you!

~Nick

closed 2023-08-29 by nmtimme