SpikeInterface/spikeinterface #1926: Suggestion for `read_cbin_ibl`

chris-langfield, 2023-08-18:
For ephys data downloaded via ONE, the AP and LFP bands are both stored in the same directory as `.cbin` files along with their associated metadata. 

Sometimes it's useful to have both cached, but this makes the following check fail (line 56) when trying to read the AP band as a recording with `read_cbin_ibl`, since there is a `cbin` for both AP and LFP:

https://github.com/SpikeInterface/spikeinterface/blob/94df77a43532fbe8671026d6aa1eb528c755a5ef/src/spikeinterface/extractors/cbin_ibl.py#L54-L59

I think it might make sense to simply change the `glob` from `"*.cbin"` to `"*.ap.cbin"`. This clarifies that we're looking for the AP band, and also doesn't fail in the case when you have LP data too.

alejoe91, 2023-08-21:
Hi @chris-langfield 

What I would do is add a `stream_name="ap"` (or "lf") in the signature, and look for `f"*.{stream_name}.cbin"` in line 55. This way, one could also read the LF stream :) Can you make a PR for it?

chris-langfield, 2023-08-21:
Sounds good! I think I need to be made a contributor I cannot push to a remote branch

alejoe91, 2023-08-21:
Can you fork and make a PR from your fork? We all use this mechanism (except for PRs to prepare a release)

chris-langfield, 2023-08-21:
Oh! Yeah of course, just never encountered that system before. I'll make a PR thanks

closed 2023-08-29 by alejoe91