SpikeInterface/spikeinterface #1927: regarding isolation score calculations

kshtjkumar, 2023-08-20:
Hi I am trying to calculate isolation score for the tetrode recording but i am not sure if I am using this command correctly or not, can someone please help :

[image]

kshtjkumar, 2023-08-20:
I am trying to achieve this : "Isolation Score of the average waveform of identified clusters in the recording. A low Isolation Score indicates the cluster of interest is close to other clusters. A high Isolation Score indicates that the cluster of interest is farther, and better separated, from other clusters."

alejoe91, 2023-08-21:
Hi @kshtjkumar 

The `mahalanobis_metrics` is a low-level function mainly used internally. I recommend using the [`compute_quality_metrics`](https://spikeinterface.readthedocs.io/en/latest/api.html#module-spikeinterface.qualitymetrics) function instead:

```
import spikeinterface.postprocessing as spost
import spikeinterface,qualitymetrics as sqm

# we is your WaveformExtractor object

pc  = spost.compute_principal_components(we) # isolation distance needs PCA

isolation = sqm.compute_quality_metrics(we, metric_names=["isolation_distance"])
```

`isolation` will be a `pandas.DataFrame` with `unit_id` as indices and an `isolation_distance` column. Let us know if this is clear!

kshtjkumar, 2023-08-22:
hi i got this:
[image]

alejoe91, 2023-08-22:
Looks good! Can I close this? Feel free to open other issues for other questions

closed 2023-08-22 by kshtjkumar