SpikeInterface/spikeinterface #1930: Problems when using export_report

claudio1212, 2023-08-22:
I have faced two problems when using the cited function: the first is the same exact problem of spikeinterface #1821, which I encountered when trying to use the "export_report" function after a sorting done using a recording that had a 3D probe attached. Is there any solution to this other than using a simpler 2D probe representation?

 Furthermore, even when using a 2D probe, the "unit localization" representation that comes out of the function seems strange: initially, axes were really extended, far more than the limits of the real array represented in mm and reported in the "csv" file I used to set the probe (which appears to be correctly defined when plotted using plot_probe). After correcting that issue, by using the same axes I used in the plot_probe to produce the units localization plot, I see that the ellipses that should represent units do not appear to be centered on the circle representing channels, as I suppose it should be. Is it normal?

alejoe91, 2023-08-24:
Hi @claudio1212 

You can instantiate a recording with a 3d probe, but some functions will need a 2d representation. You can do this inplace with the [`planarize`](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/core/baserecordingsnippets.py#L383-L404) function. Note that this function will make a 2d probe and set it in place!

claudio1212, 2023-08-24:
Thank you, I solved it using this!

closed 2023-08-25 by alejoe91