SpikeInterface/spikeinterface #1934: no probe found when running spyking circus 2, while probe attached

KLehongre, 2023-08-24:
Hello,

When I run spyking circus2 (with version 0.98.2 and 0.99.0.dev0)  I get the following warning message, while I have a probe attached to the recording.

Proof that I have a probe:

```
recordings_split[0].get_probe()
Probe - 3ch - 1shanks
```

Warning message:

```
detect peaks using locally_exclusive with n_jobs = 1 and chunk_size = 32000

detect peaks using locally_exclusive: 100%
1462/1462 [00:06<00:00, 211.82it/s]

We found 4042 peaks in total
We kept 4042 peaks for clustering
features_from_peaks with n_jobs = 1 and chunk_size = 32000

features_from_peaks: 100%
1462/1462 [00:06<00:00, 258.17it/s]

We found 6 raw clusters, starting to clean with matching...

C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\si_dev\Lib\site-packages\spikeinterface-0.99.0.dev0-py3.11.egg\spikeinterface\core\baserecordingsnippets.py:228: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn("There is no Probe attached to this recording. Creating a dummy one with contact positions")
```

Thanks!

yger, 2023-08-28:
Do you have a small test file to share to reproduce this error and the ones from the silence_period?

JoeZiminski, 2024-06-21:
Hey @KLehongre thanks for raising this issue, might you have such a a test file to share?

zm711, 2025-02-05:
I'm going to close this as stale. Feel free to reopen @KLehongre if you are able to share a small test file for us to work on :)

closed 2025-02-05 by zm711