SpikeInterface/spikeinterface #1935: spyking circus2 crashes when silence periods

KLehongre, 2023-08-24:
Hello,

I've added silence periods with "spikeinterface.preprocessing.silence_periods" to my recording to reject artefact periods, but this makes spyking circus2 crash with the following error.
I'm using version 0.98.2 of SI.

```
Error running spykingcircus2

---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In[32], line 2
      1 output_path = r"D:\01_IR-ICM\donnees\Analyses\Epilepsy\testSI\SI_pos_test_silence"
----> 2 sorting = ss.run_sorters(sorter_list, 
      3                          recordings_split, 
      4                          working_folder=output_path, 
      5                          mode_if_folder_exists="overwrite",
      6                          sorter_params=params, 
      7                          verbose=True)

File ~\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\spikeinterface\sorters\launcher.py:295, in run_sorters(sorter_list, recording_dict_or_list, working_folder, sorter_params, mode_if_folder_exists, engine, engine_kwargs, verbose, with_output, docker_images, singularity_images)
    292 if engine == "loop":
    293     # simple loop in main process
    294     for task_args in task_args_list:
--> 295         _run_one(task_args)
    297 elif engine == "joblib":
    298     from joblib import Parallel, delayed

File ~\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\spikeinterface\sorters\launcher.py:46, in _run_one(arg_list)
     43 # because we won't want the loop/worker to break
     44 raise_error = False
---> 46 run_sorter(
     47     sorter_name,
     48     recording,
     49     output_folder=output_folder,
     50     remove_existing_folder=remove_existing_folder,
     51     delete_output_folder=delete_output_folder,
     52     verbose=verbose,
     53     raise_error=raise_error,
     54     docker_image=docker_image,
     55     singularity_image=singularity_image,
     56     with_output=with_output,
     57     **sorter_params,
     58 )

File ~\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\spikeinterface\sorters\runsorter.py:148, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, **sorter_params)
    141             container_image = singularity_image
    142     return run_sorter_container(
    143         container_image=container_image,
    144         mode=mode,
    145         **common_kwargs,
    146     )
--> 148 return run_sorter_local(**common_kwargs)

File ~\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\spikeinterface\sorters\runsorter.py:176, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    174 SorterClass.run_from_folder(output_folder, raise_error, verbose)
    175 if with_output:
--> 176     sorting = SorterClass.get_result_from_folder(output_folder)
    177 else:
    178     sorting = None

File ~\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\spikeinterface\sorters\basesorter.py:289, in BaseSorter.get_result_from_folder(cls, output_folder)
    286     log = json.load(f)
    288 if bool(log["error"]):
--> 289     raise SpikeSortingError(
    290         f"Spike sorting error trace:\n{log['error_trace']}\n"
    291         f"Spike sorting failed. You can inspect the runtime trace in {output_folder}/spikeinterface_log.json."
    292     )
    294 if sorter_output_folder.is_dir():
    295     sorting = cls._get_result_from_folder(sorter_output_folder)

SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\spikeinterface\sorters\basesorter.py", line 234, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\spikeinterface\sorters\internal\spyking_circus2.py", line 73, in _run_from_folder
    recording_f = zscore(recording_f, dtype="float32")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\spikeinterface\preprocessing\normalize_scale.py", line 271, in __init__
    random_data = get_random_data_chunks(recording, **random_chunk_kwargs)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\spikeinterface\core\recording_tools.py", line 64, in get_random_data_chunks
    segment_trace_chunk = [
                          ^
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\spikeinterface\core\recording_tools.py", line 65, in <listcomp>
    recording.get_traces(
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\spikeinterface\core\baserecording.py", line 278, in get_traces
    traces = rs.get_traces(start_frame=start_frame, end_frame=end_frame, channel_indices=channel_indices)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\spikeinterface\core\channelslice.py", line 93, in get_traces
    traces = self._parent_recording_segment.get_traces(start_frame, end_frame, parent_indices)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\spikeinterface\preprocessing\silence_periods.py", line 106, in get_traces
    lower_index = np.searchsorted(self.periods[:, 1], new_interval[0])
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<__array_function__ internals>", line 200, in searchsorted
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\numpy\core\fromnumeric.py", line 1413, in searchsorted
    return _wrapfunc(a, 'searchsorted', v, side=side, sorter=sorter)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\katia.lehongre\AppData\Local\anaconda3\envs\spikeinterface\Lib\site-packages\numpy\core\fromnumeric.py", line 57, in _wrapfunc
    return bound(*args, **kwds)
           ^^^^^^^^^^^^^^^^^^^^
ValueError: object too deep for desired array

Spike sorting failed. You can inspect the runtime trace in D:\01_IR-ICM\donnees\Analyses\Epilepsy\testSI\SI_pos_test_silence\1\spykingcircus2/spikeinterface_log.json.
```

Thanks!

alejoe91, 2023-08-29:
@yger can you take a look at this? I guess there is a shape mismatch in the `silence_periods` function!

yger, 2023-08-29:
I'll have a look asap

yger, 2023-08-30:
This has been fixed with PR #1946

PR #1946 (2023-09-13): Patch `silence_periods` for mono-segment recordings
This is because when extending the dimensions of the list if only one segment, it keeps being extended when reloaded by several nodes. The patch should prevent the issues and fixes #1935
closed 2023-09-13 by alejoe91