SpikeInterface/spikeinterface #1937: Error with 'device_channel_indices' that do not match channel count in baserecordingsnippets.py

Serge-Vargo, 2023-08-25:
Hello, 

I'm running into an issue when I create my own probe object using the probeinterface package with `device_channel_indices` that exceeds the number of recording channels. For my ephys experiments, a Ripple recording system was used to record 64 uECoG electrodes using two head stages, so the .ns5 (blackrock) recording file has channel ids going from 1-32 and 129-160.     
[image]

When I manually create a probe object of my uECoG with probeinterface, I remap the physical electrodes to the actual Ripple recording channels and set the 'device_channel_indices' accordingly. 

[image]

This triggers the following error in SpikeInterface when I try to set the probe I manually created to the recording object 
[image]
[image]
[image]

 When looking at the source code where this error gets flagged, I can see there's a TODO comment, so perhaps this is something you are currently looking at. I just wanted to bring it up here as well. Hopefully this is an easy fix?

[image]

My intuition tells me that line 146 of baserecordingsnippets.py could be replaced with the len() function instead of the np.max() function, but I'm not sure how that would affect the edge case of loading a probe with no channel list. 

I'm able to work around this issue when I set the `device_channel_indices` using probeinterface with `list(range(0,64))`, so I'm able to run the rest of the pipeline. It'll be nice to have a one-to-one map of physical electrodes to recording channel number tho. 

Thanks,
Sergio

samuelgarcia, 2023-08-28:
Hi @Serge-Vargo
you have to handle `el_num_scheme` as indices. In short a int number zero based in 0-63(included) range.
In your code `el_num_scheme` is the ID (transformed as an integer) which is wrong.

Serge-Vargo, 2023-08-30:
Hi @samuelgarcia,
Thanks for looking into this. Understood. I can handle `el_num_scheme` as indices, not a problem.

closed 2023-08-30 by Serge-Vargo