SpikeInterface/spikeinterface #1938: read_plexon() ValueError: When changing to a larger dtype, its size must be a divisor of the total size in bytes of the last axis of the array

shunnnli, 2023-08-25:
Hi,

We're loading a .plx file using read_plexon() method. We know the code runs because a sample short (~1min) .plx file (acquired directly in .plx) runs in our code successfully.

However, when we try to load the actually file (~40min recording in .pl2 format, converted to .plx), spikeinterface returns following error (see below).

Thanks in advance!

**My code to read plexon file:**
`raw_rec = si.read_plexon(recording_path,stream_name='Signals 0')`

**Spikeinterface error:**
> ValueError                                Traceback (most recent call last)
Cell In[11], line 2
      1 # we do not load the sync channel, so the probe is automatically loaded
----> 2 raw_rec = si.read_plexon(recording_path,stream_name='Signals 0')
      3 raw_rec

File ~\.conda\envs\spikeinterface\lib\site-packages\spikeinterface\core\core_tools.py:26, in define_function_from_class.<locals>.reader_func(*args, **kwargs)
     24 @copy_signature(source_class)
     25 def reader_func(*args, **kwargs):
---> 26     return source_class(*args, **kwargs)

File ~\.conda\envs\spikeinterface\lib\site-packages\spikeinterface\extractors\neoextractors\plexon.py:29, in PlexonRecordingExtractor.__init__(self, file_path, stream_id, stream_name, all_annotations)
     27 def __init__(self, file_path, stream_id=None, stream_name=None, all_annotations=False):
     28     neo_kwargs = self.map_to_neo_kwargs(file_path)
---> 29     NeoBaseRecordingExtractor.__init__(self, stream_id=stream_id, 
     30                                        stream_name=stream_name,
     31                                        all_annotations=all_annotations, 
     32                                        **neo_kwargs)
     33     self._kwargs.update({'file_path': str(file_path)})

File ~\.conda\envs\spikeinterface\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py:42, in NeoBaseRecordingExtractor.__init__(self, stream_id, stream_name, block_index, all_annotations, **neo_kwargs)
     39 def __init__(self, stream_id=None, stream_name=None, 
     40              block_index=None, all_annotations=False, **neo_kwargs):
---> 42     _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
     44     kwargs = dict(all_annotations=all_annotations)
     45     if block_index is not None:

File ~\.conda\envs\spikeinterface\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py:24, in _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
     23 def __init__(self, block_index, **neo_kwargs):
---> 24     self.neo_reader = get_reader(self.NeoRawIOClass, **neo_kwargs)
     26     if self.neo_reader.block_count() > 1 and block_index is None:
     27         raise Exception("This dataset is multi-block. Spikeinterface can load one block at a time. "
     28                         "Use 'block_index' to select the block to be loaded.")

File ~\.conda\envs\spikeinterface\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py:14, in get_reader(raw_class, **neo_kwargs)
     12 neoIOclass = eval('neo.rawio.' + raw_class)
     13 neo_reader = neoIOclass(**neo_kwargs)
---> 14 neo_reader.parse_header()
     16 return neo_reader

File ~\.conda\envs\spikeinterface\lib\site-packages\neo\rawio\baserawio.py:179, in BaseRawIO.parse_header(self)
    166 def parse_header(self):
    167     """
    168     This must parse the file header to get all stuff for fast use later on.
    169 
   (...)
    177 
    178     """
--> 179     self._parse_header()
    180     self._check_stream_signal_channel_characteristics()

File ~\.conda\envs\spikeinterface\lib\site-packages\neo\rawio\plexonrawio.py:90, in PlexonRawIO._parse_header(self)
     88 pos = offset4
     89 while pos < data.size:
---> 90     bl_header = data[pos:pos + 16].view(DataBlockHeader)[0]
     91     length = bl_header['NumberOfWaveforms'] * bl_header['NumberOfWordsInWaveform'] * 2 + 16
     92     bl_type = int(bl_header['Type'])

ValueError: When changing to a larger dtype, its size must be a divisor of the total size in bytes of the last axis of the array.

alejoe91, 2023-08-29:
Thanks for reporting this @shunnnli 

Note that we'll have a plexon2 reader very soon which might fix the issue: https://github.com/SpikeInterface/spikeinterface/pull/1736

If you have time, you can try to checkout that PR and beta test it :)

alejoe91, 2023-11-06:
@shunnnli were you able tot try the `plexon2` reader?

zm711, 2024-03-21:
Just want to bumpythis one more time. @shunnnli, did the plexon2 reader work for you ?

zm711, 2024-12-03:
I'll close this for inactivity. We have worked on Plexon1 and Plexon2 over on the Neo side. A Plexon engineer is helping update some functions at Neo. Feel free to open an issue over there if you run into more issues with plexon files so we can track there.

closed 2024-12-03 by zm711

ChauhanT, 2024-12-06:
Hi @zm711 , I am reading .plx files and am unable to read events from it. Using neo I can read stream_ids like EVT01 for events (digital in) and 0, 1, etc for analogue data. spikeinterface seems to only read analogue data. Am I missing something here ? 

Thank you!