SpikeInterface/spikeinterface #1939: how to install the old  version of spikeinterface ,such as 0.96.0?

li1901, 2023-08-27:
how to install the old  version of spikeinterface ,such as 0.96.0?
Cause i need to use a baned moudle named spikeinterface.toolkit due to my seniors‘s code.

florian6973, 2023-08-28:
Hi! You can try `pip install spikeinterface==0.96.0`

alejoe91, 2023-08-29:
Thanks for replying @florian6973 

@li1901 I would recommend to refactor the code to be able to use the latest versions. Basically we further split the `toolkit` module into:
- `preprocessing`
- `postprocessing`
- `qualitymetrics`

So you just need to change the `spikeinterface.toolkit` functions to the appropriate submodule. For example:
```
spikeinterface.toolkit.preprocessing --> spikeinterface.preprocessing
```
and so on!

Closing this for now, feel free to open another issue if you have other problems!

closed 2023-08-29 by alejoe91