SpikeInterface/spikeinterface #1942: SortingExtractor.get_total_num_spikes() returns units with 0 spikes after sorting

steevelaquitaine, 2023-08-28:
I run Kilosort3 on this dataset  'sub-MEAREC-250neuron-Neuropixels_ecephys.nwb' available from [Matthias Hennig's tutorial](http://homepages.inf.ed.ac.uk/mhennig/)](https://spikeinterface.github.io/blog/ground-truth-comparison-and-ensemble-sorting-of-a-synthetic-neuropixels-recording/)

The `get_total_num_spikes()` method of the resulting `SortingExtractor` is returning some units with 0 spikes (notice  unit 8 below). I can't make sense of this. How can Kilosort3 create unit clusters without spikes?

```
Sorting = si.load_extractor(sorting_path)
total_spike_per_unit = Sorted.get_total_num_spikes()
print(total_spike_per_unit)
```

{0: 2279,
 1: 2846,
 2: 2935,
 3: 2246,
 4: 213,
 5: 348,
 6: 196,
 7: 273,
 8: 0,
 9: 125,
...

I am running `spikeinterface==0.96.1`

samuelgarcia, 2023-08-29:
Hi Steeve.
The question should be ask to autor of kilosort but here a possible explanation.
kilosort use internally template matching which have 2 steps: 1. discover a catalogue of possible template 2. assign every spike to a template.
Maybe in step 1 two very similar templates are "discoverd" and in step 2 all spikes could be assigned to the first one only.

We have a function to remove then : https://spikeinterface.readthedocs.io/en/latest/api.html?highlight=empty#spikeinterface.core.BaseSorting.remove_empty_units

steevelaquitaine, 2023-08-29:
Thank you for your quick reply and explanation! I will then confirm with the authors of Kilosort.
Best,
Steeve

alejoe91, 2023-08-29:
Perfect, closing this! Feel free to open another issue for other questions ;)

closed 2023-08-29 by alejoe91