SpikeInterface/spikeinterface #1945: Export to Phy : getting ValueError: attempt to get argmax of an empty sequence

mandarmp, 2023-08-30:
Try to export to phy and visualize.

`from spikeinterface.postprocessing import compute_spike_amplitudes, compute_principal_components
from spikeinterface.exporters import export_to_phy

# the waveforms are sparse so it is faster to export to phy
folder = '/home/mmp/disktb/mmpatil/codbase/MEA_Analysis/SpikeSortingPipeline/sorting_intermediate/waveforms/'

we = si.extract_waveforms(recording_chunk,sorting_good,folder=folder,sparse=True)
# some computations are done before to control all options
compute_spike_amplitudes(we)
compute_principal_components(we, n_components=3, mode='by_channel_global')

# the export process is fast because everything is pre-computed
export_to_phy(we, output_folder='path/to/phy_folder')`

After this , running the phy command 

`phy template-gui  /mnt/disk15tb/mmpatil/codbase/MEA_Analysis/SpikeSortingPipeline/path/to/phy_folder/params.py`

Leads to the error:

`16:18:13.711 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
16:18:14.509 [E] __init__:62          An error has occurred (ValueError): attempt to get argmax of an empty sequence
Traceback (most recent call last):
  File "/home/mmp/.local/bin/phy", line 8, in <module>
    sys.exit(phycli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/mmp/.local/lib/python3.10/site-packages/phy/apps/__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "/home/mmp/.local/lib/python3.10/site-packages/phy/apps/template/gui.py", line 218, in template_gui
    gui = controller.create_gui()
  File "/home/mmp/.local/lib/python3.10/site-packages/phy/apps/base.py", line 1700, in create_gui
    self.supervisor.attach(gui)
  File "/home/mmp/.local/lib/python3.10/site-packages/phy/cluster/supervisor.py", line 950, in attach
    self._create_views(
  File "/home/mmp/.local/lib/python3.10/site-packages/phy/cluster/supervisor.py", line 765, in _create_views
    gui, data=self.cluster_info, columns=self.columns, sort=sort)
  File "/home/mmp/.local/lib/python3.10/site-packages/phy/cluster/supervisor.py", line 925, in cluster_info
    return [self.get_cluster_info(cluster_id) for cluster_id in self.clustering.cluster_ids]
  File "/home/mmp/.local/lib/python3.10/site-packages/phy/cluster/supervisor.py", line 925, in <listcomp>
    return [self.get_cluster_info(cluster_id) for cluster_id in self.clustering.cluster_ids]
  File "/home/mmp/.local/lib/python3.10/site-packages/phy/cluster/supervisor.py", line 750, in get_cluster_info
    out[key] = func(cluster_id)
  File "/home/mmp/.local/lib/python3.10/site-packages/phy/apps/base.py", line 1194, in get_best_channel_label
    return self._get_channel_labels([self.get_best_channel(cluster_id)])[0]
  File "/home/mmp/.local/lib/python3.10/site-packages/phy/utils/context.py", line 154, in memcached
    out = f(*args, **kwargs)
  File "/home/mmp/.local/lib/python3.10/site-packages/phy/apps/base.py", line 1188, in get_best_channel
    channel_ids = self.get_best_channels(cluster_id)
  File "/home/mmp/.local/lib/python3.10/site-packages/phy/utils/context.py", line 154, in memcached
    out = f(*args, **kwargs)
  File "/home/mmp/.local/lib/python3.10/site-packages/phy/apps/template/gui.py", line 151, in get_best_channels
    template = self.model.get_template(template_id)
  File "/home/mmp/.local/lib/python3.10/site-packages/phylib/io/model.py", line 957, in get_template
    return self._get_template_sparse(template_id, unwhiten=unwhiten)
  File "/home/mmp/.local/lib/python3.10/site-packages/phylib/io/model.py", line 924, in _get_template_sparse
    best_channel = channel_ids[np.argmax(amplitude)]
  File "<__array_function__ internals>", line 200, in argmax
  File "/home/mmp/.local/lib/python3.10/site-packages/numpy/core/fromnumeric.py", line 1242, in argmax
    return _wrapfunc(a, 'argmax', axis=axis, out=out, **kwds)
  File "/home/mmp/.local/lib/python3.10/site-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return bound(*args, **kwds)
ValueError: attempt to get argmax of an empty sequence`

Any guidance/ suggestion regarding this is greatly appreciated. Thank you for your time

alejoe91, 2023-08-30:
Hi @mandarmp 

How many channels do you have? 

Maybe you are using a default sparsity which is too narrow for your electrode configurations, resulting in mono-channel templates (which might not be dealt properly at export or in Phy). This is only a guess though!
Could you check that the sparse templates have a correct shape?

```
for unit_id in we.unit_ids:
    template = we.get_template(unit_id)
    print(f"Unit {unit_id} - template shape: {template.shape}")
```

mandarmp, 2023-08-30:
Hello @alejoe91 !

Thank you for your reply. I get what the issue is now. There are a few mono-channel templates, since this recording is a Maxwell Network activity with 1020 high spiking channels which make the template too sparse. May be.

closed 2023-08-30 by mandarmp