SpikeInterface/spikeinterface #1947: Support for Brainwave/Biocam 4.x file format

vdtang3, 2023-08-30:
I'm trying to process some data that has been acquired using a Biocam acquisition system, but it seems like the formatting/metadata is quite different what is currently supported by spikeinterface. Are there plans to add extractor support for the latest Biocam 4.x format in the near future? Thanks!

alejoe91, 2023-09-01:
Hi, do you have some documentation on the format and some small test files?

All the I/O is handled through the [NEO](https://github.com/NeuralEnsemble/python-neo) package. Can you open an issue there?

vdtang3, 2023-09-01:
Just sent the documentation via email, and opened up an issue on NEO. Thanks!

JoeZiminski, 2024-06-21:
Closing as issue opened on the Neo side, @vdtang3 please feel free to reopen if once implemented on the NEO side anything on the SpikeInterface side is causing you problems.

closed 2024-06-21 by JoeZiminski