SpikeInterface/spikeinterface #1949: NotImplementedError: This meta file version of spikeglxis not implemented spikinterface 0.98.1

gmaggi, 2023-08-31:
Hi,

One of the users of spikeinterface has a single recording which is in a directory together to its metafile.. Te full content of the dir is: `sh0br80bvd_g0_tcat.imec0.ap.bin  sh0br80bvd_g0_tcat.imec0.ap.meta  sh0br80bvd_g0_tcat.imec0.lf.bin  sh0br80bvd_g0_tcat.imec0.lf.meta`

we are running `se.read_spikeglx( DIRPATH , stream_id = 'imec0.ap')`
and we get (last part of the error)

`File "/opt/anaconda/anaconda3/envs/spikeinterfacefull_0.98.1/lib/python3.10/site-packages/neo/rawio/spikeglxrawio.py", line 406, in extract_stream_info
    raise NotImplementedError('This meta file version of spikeglx'
NotImplementedError: This meta file version of spikeglxis not implemented`

any idea what is the issue could be?

thanks!

alejoe91, 2023-08-31:
Can you share the `.meta` file?

gmaggi, 2023-08-31:
I just sent it to your gmail. Sorry if this is not the right way of sharing it

alejoe91, 2023-08-31:
Thanks!  It works on my side! What version of neo are you using? Can you `pip install -U neo` to be sure?

gmaggi, 2023-08-31:
python -c "import neo; print(neo.__version__)"
0.12.0

that is the current one, do I still need to update?

gmaggi, 2023-08-31:
neo 0.12.0 seems to be the latest one. The update did not fix the issue.
The  spikeinterface we have is 0.98.1

alejoe91, 2023-08-31:
Can you paste the full error?

alejoe91, 2023-08-31:
Ah! I think that the problem is that it's a NP2 probe (imDatPrb_type=24), but you have LF streams too! NP2 probes should only generate AP streams, since they are broad-band...

Can you confirm you are using a NP2-4 shank probe?

Shrh627, 2023-08-31:
Hi,
Yes, we are using a 4-shank NP2.0 probe. I also need to mention that I found a difference between the raw data meta file and the output of the CatGT. 

catGTCmdline0=<CatGT -dir=E:/Data/ephys/raw/Chronic/SH049/230823/sh0br80bvd_g0/sh0br80bvd_g0_imec0 -run=sh0br80bvd -g=0 -t=0 -no_run_fld -t_miss_ok -prb=0 -ap -lf -aphipass=300 -aplopass=5000 -lfhipass=1 -lflopass=300 -gbldmx -gfix=0.4,0.1,0.02 -dest=//nerffs17/boninlabwip2023/data/ephys/shahriar/Raw/NP2.0/Chronic/SH049/230823>
catGVals=0,0
catNFiles=1
catTVals=0,0

These lines are in the output of the CatGT .meta file which is not in the GLX output file.

alejoe91, 2023-08-31:
Do you have `lf.bin` in the original folder (before CatGT)?

Shrh627, 2023-08-31:
No, I don't.

alejoe91, 2023-08-31:
Then try to move them somewhere else! It should work! We made that assumption that NP2 only has ap, but if CatGT produces an lf too we'll probably need to relax thta assumption

gmaggi, 2023-09-01:
Putting only the ap* files in a dir works fine. 
Thanks!!

gmaggi, 2023-09-27:
> Then try to move them somewhere else! It should work! We made that assumption that NP2 only has ap, but if CatGT produces an lf too we'll probably need to relax thta assumption

Hi @alejoe91 , would this assumption be relaxed in the next version?

alejoe91, 2023-09-27:
Yes: see https://github.com/NeuralEnsemble/python-neo/pull/1327

alejoe91, 2023-11-06:
Closing because related to NEO (and already fixed in SI)

closed 2023-11-06 by alejoe91