SpikeInterface/spikeinterface #1956: Issue with TdtRecordingExtractor and spykingcircus (get_shared_channel_property_names)

bradlytstone, 2023-09-02:
I have implemented the following pipeline to sort my TdT file. However, while this used to work, it no longer does, and I have tried several things: upgrading/downgrading spikeinterface, manually putting in property names. I am just lost as to how to fix this, which has been an arduous process just getting the TdT file to properly be sorted (when it used to work). I am electing to use spykingcircus because this is what we previously got to work (and no other sorters did). Can you provide assistance please?  

# Import functions
import spikesorters as ss
import spikeinterface as sii
import spikeinterface.preprocessing as spre
from spikeinterface import extractors as se

# Read in tdt file
reader =se.read_tdt('folder',stream_name='stream')

# Design the probe based on experiment parameters
probe = generate_multi_columns_probe(num_columns=2, xpitch=43, ypitch=50,
    num_contact_per_column=[16, 16], y_shift_per_column=[0, 25],
    contact_shapes='circle', contact_shape_params={'radius': 7.5})

# Channel mappings
device_channel_indices = [ 7, 5, 3, 1, 30, 28, 25, 27, 6, 29, 4, 31, 2, 24, 0, 26, 9, 11,
    13, 15, 16, 18, 23, 21, 8, 19, 10, 17, 12, 22, 14, 20]

# Attach the probe to the recording device using the channel mappings
probe.set_device_channel_indices(device_channel_indices)

# Set probe within reader object
reader = reader.set_probe(probe)

# Parameters for filtering and saving the recording
params = {'chunk_duration': '1s', 'common_ref_removal': False, 
    'detect_sign': -1,'detect_threshold': 7, 'freq_max': 3000.0, 
    'freq_min': 300.0, 'n_jobs': 10,'nested_params': None, 
    'progress_bar': True, 'chunk_size': 5000}

save_kwargs = {'n_jobs': 10, 'chunk_size': 20000}

# Apply a bandpass filter to the data
reader_filtered = spre.bandpass_filter(reader, freq_min=params['freq_min'], freq_max=params['freq_max'])

# Detect and remove bad channels
bad_channel_ids, channel_labels = spre.detect_bad_channels(reader_filtered, noisy_channel_threshold=0.4)
rec_clean = reader.remove_channels(bad_channel_ids)

# Update channel_id map based on the cleaned data
clean_dev_channels, clean_id_channels = zip(*((x, y) for x, y in \
                zip(device_channel_indices, reader_filtered.channel_ids)\
                    if y in rec_clean.channel_ids))

# Apply a common median reference (CMR) to the data
recording_cmr = spre.common_reference(rec_clean, reference='global', operator='median')

# Indicate that the recording has been filtered
recording_cmr.annotate(is_filtered=True)

# Set sorting params
params = {  'adjacency_radius': adjacency_radius,
    'detect_threshold': detect_threshold,
    'auto_merge': auto_merge,
    'num_workers': internal_workers}

out = ss.sorters.run_spykingcircus(recording = recording_cmr, **params,\
                     output_folder='spykcirc2_output_v4')

# Error
AttributeError: 'TdtRecordingExtractor' object has no attribute 'get_shared_channel_property_names'

bradlytstone, 2023-09-07:
Hello again. So, having followed the same code sequence as pasted above, with the changes you mentioned, I am getting a new error in the JSON file: 

# Error
{
    "sorter_name": "spykingcircus",
    "sorter_version": "1.1.0",
    "datetime": "2023-09-07T18:14:42.261104",
    "runtime_trace": [],
    "error": true,
    "error_trace": "Traceback (most recent call last):\n  File \"C:\\Users\\SuperComputer1\\miniconda3\\lib\\site-packages\\spikeinterface-0.98.0.dev0-py3.9.egg\\spikeinterface\\sorters\\basesorter.py\", line 226, in run_from_folder\n    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)\n  File \"C:\\Users\\SuperComputer1\\miniconda3\\lib\\site-packages\\spikeinterface-0.98.0.dev0-py3.9.egg\\spikeinterface\\sorters\\spyking_circus\\spyking_circus.py\", line 165, in _run_from_folder\n    raise Exception('spykingcircus returned a non-zero exit code')\nException: spykingcircus returned a non-zero exit code\n",
    "run_time": null
}

Do you have suggestions on how to fix this? Sorry for the continued troubles.

bradlytstone, 2023-11-04:
I am pumping this up, as we are still experiencing this issue.

alejoe91, 2023-11-04:
Can you try to update to the latest release (from yesterday)?

`pip install -U spikeinterface`

SpykingCircus 2 underwent a big update too :)