SpikeInterface/spikeinterface #1959: Refactor tests

samuelgarcia, 2023-09-04:
Since the new generate.py is in the main branch, we should refactor some tests:
  * use as much as possble the new generate_ground_truth_recording() with the appropriate channel size and duration
  * remove toy_example() in tests globally but keep in examples.
  * avoid the downloading of the mearec and use the new generator
  * avoid saving to disk recording/sorting/waveforms extractors when not necessary the waveforms in memory works weel 
     with the lazy generator normally.
  * check that we have seed when generating something. Many generate_sorting() do not have!
  * avoid generating several time the same object in a module (even this is now almost lazy).

@alejoe91 
@h-mayorquin

h-mayorquin, 2023-09-04:
Yes, this looks good.

Tests for most processing / functionality should be done in memory and quick to make the full tests suit fast. Then IO (calls to .save on objects) should be tested separatley. We decouple and make our testing suite more efficiently. Both good things at the costs of refactor.

I did not have time to review the refactoring of the generator module. I will take a look at the other points and take a look to the status of main branch tomorrow. Hopefully before our meeting.

zm711, 2024-06-19:
I think this was mostly accomplished in the hackathon no? I'll close this for now, but if I'm mistaken or misunderstood just reopen.

closed 2024-06-19 by zm711