SpikeInterface/spikeinterface #1960: se.read_openephys ValueError: invalid literal for int() spikeinterface 0.98.1

gmaggi, 2023-09-05:
Hi,

I was testing the extractor for a spikeinterface user. I got this error

```
In [6]: se.read_openephys(Path("/mnt/takeokalab/takeokalabwip2023/JJ/20230706/"))
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[6], line 1
----> 1 se.read_openephys(Path("/mnt/takeokalab/takeokalabwip2023/JJ/20230706/"))
```

And at the end:

```
File /opt/anaconda/anaconda3/envs/spikeinterfacefull_0.98.1/lib/python3.10/site-packages/neo/rawio/openephysbinaryrawio.py:481, in explore_folder(dirname, experiment_names)
    479 experiment_folder = root.parents[0]
    480 experiment_name = experiment_folder.stem
--> 481 experiment_id = int(experiment_name.replace('experiment', ''))
    482 if experiment_name not in possible_experiment_names:
    483     possible_experiment_names.append(experiment_name)

ValueError: invalid literal for int() with base 10: 'JJ'
```

The content of the dir is:

 ```
ls /mnt/takeokalab/takeokalabwip2023/JJ/20230706
continuous.dat  continuous.nrs  continuous.xml  sample_numbers.npy  structure.oebin
```

Any idea what could be?
Thanks in advance!

alejoe91, 2023-09-05:
Yes. The open ephys outputs a folder structure as follows:

`record node -> experiment{experimentID} -> recording{recordingID} -> continuous`

The `structure.oebin` is in the `recording{recordingID}`, the `.dat` and `.npy` files in the `continuous`.
The folder that you have is a mix and has been probably manually modified so it's not a valid Open Ephys folder, unfortunately.

gmaggi, 2023-09-05:
Thank you!. I am waiting for the user feedback. Feel free to close this ticket.