SpikeInterface/spikeinterface #1961: get_default_params in 0.96.0 is not present in 0.98.1

gmaggi, 2023-09-05:
Hi, 

spikeinterface.sorters 0.96.0 had a convenient function to check the parameters that a user was passing. https://spikeinterface.readthedocs.io/en/0.96.0/getting_started/plot_getting_started.html#getting-started-tutorial

We used to check the keys of `get_default_params` and compare with was the user was passing to.

It seems in 0.98.1 `get_default_params` is no longer present. any similar functionality?

zm711, 2023-09-05:
@gmaggi 
You're right the function name has changed. There are two functions now:

`get_default_sorter_params` to see defaults. If you need the descriptions of these parameters you can run `get_sorter_params_descriptions`.

For example:
```python
si.get_default_sorter_params('tridesclous')
```
will return a dictionary of the default params. I'll put in a PR to fix that doc now.

gmaggi, 2023-09-05:
Thank you, that works!

closed 2023-09-05 by gmaggi