SpikeInterface/spikeinterface #1964: Read the docs structure structure nests core tutorials inside widget tutorials

h-mayorquin, 2023-09-06:
If I clik on the module examples gallery, then core tutorials and finally append and concatenate segments the side menu makes it appears as if this embedded in the widgets tutorials.

[image]

Not sure why this is the case or if this indicates a deeper error in the structure of our documentation.

zm711, 2023-09-06:
@h-mayorquin,
I think it's been like that for ages, someone on an issue recently shared the docs for .96. It was not nested on version 0.13 though, so between 13-96 api change created the nested structure in the docs. I'll look around a bit too.
https://spikeinterface.readthedocs.io/en/0.96.0/modules/index.html

zm711, 2023-09-06:
I wonder if we change the order of the `modules_gallery` so that widgets isn't last, but lets say core is last if the mistake is the same. I looked at the raw build log and nothing looks weird. It must be accidentally creating the nested structure in the `index.rst` of widgets, but I don't have a clue why. The `conf.py` looks fine.

zm711, 2023-09-14:
Since I wasn't able to get this fixed locally (soon-ish) I might open a do not merge pull request where I try a bunch of rearranging things to see if I can isolate the issue in the rtd building process.

zm711, 2023-09-22:
Fixed in #2036 
https://spikeinterface.readthedocs.io/en/latest/modules_gallery/index.html

closed 2023-09-22 by zm711

h-mayorquin, 2023-09-23:
This is great, thanks a bunch.