SpikeInterface/spikeinterface #1967: OpenEphysBinaryRecordingExtractor' object has no attribute 'load_probe_file' spikeinterface 0.98.1

gmaggi, 2023-09-06:
Hi, we got this error:

```
readephys = se.read_openephys(Path("/mnt/takeokalab/takeokalabwip2023/JJ/2022-05-25_18-33-05/"))

In [5]: print(readephys)
OpenEphysBinaryRecordingExtractor: 32 channels - 30.0kHz - 1 segments - 11,198,100 samples 
                                   373.27s (6.22 minutes) - int16 dtype - 683.48 MiB

In [6]: readephys.load_probe_file("/home/jyhjangs-takeokalab/CambridgeNT_H5_CB.prb")
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[6], line 1
----> 1 readephys.load_probe_file("/home/jyhjangs-takeokalab/CambridgeNT_H5_CB.prb")

AttributeError: 'OpenEphysBinaryRecordingExtractor' object has no attribute 'load_probe_file'

In [7]: ls /mnt/takeokalab/takeokalabwip2023/JJ/2022-05-25_18-33-05/
'Record Node 110'/  'Record Node 1104.375s - 4.5.svg'*
```

any idea what we are doing wrong ?

alejoe91, 2023-09-06:
I don't think there has ever been a `load_prb_file`! You should load the prb file with `probeinterface.read_prb()` and use the `set_probe` function

gmaggi, 2023-09-06:
Ok, any tutorial for this. I got an error with set_probe, then I tried set_probeS and got another kind of error:

```
readephys = se.read_openephys(Path("/mnt/takeokalab/takeokalabwip2023/JJ/2022-05-25_18-33-05/"))
In [10]: prb = read_prb("/home/jyhjangs-takeokalab/CambridgeNT_H5_CB.prb")

In [11]: readephys.set_probe(prb)
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
Cell In[11], line 1
----> 1 readephys.set_probe(prb)

File /opt/anaconda/anaconda3/envs/spikeinterfacefull_0.98.1/lib/python3.10/site-packages/spikeinterface/core/baserecordingsnippets.py:77, in BaseRecordingSnippets.set_probe(self, probe, group_mode, in_place)
     73 def set_probe(self, probe, group_mode="by_probe", in_place=False):
     74     """
     75     Wrapper on top on set_probes when there one unique probe.
     76     """
---> 77     assert isinstance(probe, Probe), "must give Probe"
     78     probegroup = ProbeGroup()
     79     probegroup.add_probe(probe)

AssertionError: must give Probe

In [12]: readephys.set_probes(prb)
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[12], line 1
----> 1 readephys.set_probes(prb)

File /opt/anaconda/anaconda3/envs/spikeinterfacefull_0.98.1/lib/python3.10/site-packages/spikeinterface/core/baserecordingsnippets.py:154, in BaseRecordingSnippets.set_probes(self, probe_or_probegroup, group_mode, in_place)
    147 if number_of_device_channel_indices >= self.get_num_channels():
    148     error_msg = (
    149         f"The given Probe have 'device_channel_indices' that do not match channel count \n"
    150         f"{number_of_device_channel_indices} vs {self.get_num_channels()} \n"
    151         f"device_channel_indices are the following: {device_channel_indices} \n"
    152         f"recording channels are the following: {self.get_channel_ids()} \n"
    153     )
--> 154     raise ValueError(error_msg)
    156 new_channel_ids = self.get_channel_ids()[device_channel_indices]
    157 probe_as_numpy_array = probe_as_numpy_array[order]

ValueError: The given Probe have 'device_channel_indices' that do not match channel count 
63 vs 32 
device_channel_indices are the following: [ 0  1  2  3  4  5  6  7  8  9 10 11 12 13 14 15 16 17 18 19 20 21 22 23
 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47
 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63] 
recording channels are the following: ['CH1' 'CH2' 'CH3' 'CH4' 'CH5' 'CH6' 'CH7' 'CH8' 'CH9' 'CH10' 'CH11'
 'CH12' 'CH13' 'CH14' 'CH15' 'CH16' 'CH17' 'CH18' 'CH19' 'CH20' 'CH21'
 'CH22' 'CH23' 'CH24' 'CH25' 'CH26' 'CH27' 'CH28' 'CH29' 'CH30' 'CH31'
 'CH32'] 

In [13]: 

```

alejoe91, 2023-09-06:
It seems that the probe has 64 channels, the recording 32. They need to match 8n order.to be able to load the probe

gmaggi, 2023-09-06:
Ok, I'll pass the message to the user. 
BTW, do we have to use set_probeS, no set_probe, correct?

alejoe91, 2023-09-06:
If you have one probe with all channels you can also use `set_probe`. You may also use a probe group! Check out the probeinterface docs ;)

h-mayorquin, 2023-09-06:
@gmaggi was the first code suggestion generated with a LLM? : P

h-mayorquin, 2023-09-06:
I am saying that maybe one of your end users probably used an LLM (like ChatGPT) to generate the piece of code that produced the first error you have shown.

gmaggi, 2023-09-27:
I think we can close this

closed 2023-09-27 by gmaggi