SpikeInterface/spikeinterface #1975: how to use S.I. with custom data in MountainSort 5.

li1901, 2023-09-11:
We have obtained a small dataset in MATLAB, which is of the format M*N (where M represents the number of channels, and N represents the number of timestamp points) using custom probes. Now, we would like to sort our data using MountainSort 5. However, I am encountering problems while using SpikeInterface to transform our private data into the correct format, as mentioned in the MountainSort 5 README:

`recording = ... # Load your recording using SpikeInterface`

Do you have any guide documents or examples for this?
thank u so much.

h-mayorquin, 2023-09-11:
Hi, @li1901 
Hope we can help you. Some information would be useful in that directrion.
How small is your data, that is, how large are M and N? What does it mean that you have in matlab? Is it in the matlab format?

There was an issue recently were we discussed how to load data from matalab but I could not find it. Any chance @alejoe91 or @samuelgarcia remember the title?

li1901, 2023-09-11:
> 
You're welcome for your reply. My MATLAB data is currently in a variable in the workspace, with a size of 16*1,000,000. The sampling rate of my data is 20,000 Hz.  I can convert or save it in a different format through MATLAB, for example, saving it as a .mat.

li1901, 2023-09-14:
i have soleved this by using trans my data to mda-format, and read it by spikeinterface.ex....

h-mayorquin, 2023-09-14:
That's great. 

Would you mind sharing a small script on how you did this in case some user encounters this thread in the future? It could be useful for someone else.

Sorry for not answering before, this has been a busy week for us. I was actually thinking on a solution along these lines which I will left here for future reference:

https://www.mathworks.com/matlabcentral/fileexchange/58975-mat2np-turn-matlab-array-into-a-python-numpy-object

alejoe91, 2023-09-14:
> Hi, @li1901 Hope we can help you. Some information would be useful in that directrion. How small is your data, that is, how large are M and N? What does it mean that you have in the data matlab, Is it in the matlab format (.m)?
> 
> There was an issue recently were we discussed how to load data from matalab but I could not find it. Any chance @alejoe91 or @samuelgarcia remember the title?

@h-mayorquin I think we discussed about making an exporter to `.mat` of a Sorting object. We could also have a very simple `MatlabRecording`  that can load a .mat file with 2D data in it to facilitate this.

zm711, 2023-09-19:
#2018 
Just linking this to the PR with the documentation changes.

h-mayorquin, 2023-09-28:
Closing this, we have a section on how to do this:

https://spikeinterface.readthedocs.io/en/latest/how_to/load_matlab_data.html

If any encounters this on the future and does not understand the how to guide or has some feedback please open a new issue and we will be happy to clarify or discuss.

closed 2023-09-28 by h-mayorquin