SpikeInterface/spikeinterface #1978: Create `Templates` dataclass

h-mayorquin, 2023-09-11:
Simple `dataclass` for templates, including:
- ms_before/after
- nbefore/after
- channel_ids/unit_ids
- channel_locations
- sparsity

Needs to be picklable and JSON-serializable

CodyCBakerPhD, 2023-09-11:
> Needs to be picklable and JSON-serializable

Going with a Pydantic Model would go a long way beyond a traditional `dataclass` towards helping with these, as well as auto-generation of the accompanying JSON schema (not to mention native type checking / input validation)

alejoe91, 2023-09-11:
Yeah for the first round this will be mainly used internally, so I think that pydantic might bean overkill. Let's keep this in mind for future developments.

h-mayorquin, 2023-09-20:
#1982

closed 2023-11-15 by alejoe91