SpikeInterface/spikeinterface #2007: Problem when reading neuralynx header from Cheetah 5.6.2

JohnStout, 2023-09-18:
Using the spikeinterface.extractors.read_neuralynx does not seem to work on headers created using Cheetah software 5.6.2. Works fine on header from Cheetah 6.4.1.

Here is a link to google drive with two CSC.ncs files (one that works with read_neuralynx and one that doesn't work for me) https://drive.google.com/drive/folders/1RRVdPkOuQY946adaJOEH6Sp0w0XwmISh?usp=share_link

Attached is an image of the error.

[image]

zm711, 2023-09-18:
@JohnStout 

Looks like there is an open PR at https://github.com/NeuralEnsemble/python-neo/pull/1257 (and associated issue https://github.com/NeuralEnsemble/python-neo/issues/1202) trying to deal with Cheetah 5.6.x. One of the problems is that they need test files that could be included in the public test repository to ensure the fixes work at the Neo level. If you'd be willing to share your failing file as part of the test suite it may facilitate getting that PR merged, which then could be propagated to SpikeInterface.

JohnStout, 2023-09-19:
> Looks like there is an open PR at NeuralEnsemble/python-neo#1257
> issue NeuralEnsemble/python-neo#1202

Hey Zach,

Did you receive those google drive files? Two example files, one with newer
version of cheetah. One with older. They're just folders with CSC files in
them.

Do you happen to know of any short term solutions?

Thanks!
John

zm711, 2023-09-19:
https://spikeinterface.readthedocs.io/en/latest/installation.html

Goes through how to install spikeinterface and neo from source. But in your case you would need to use the branch of neo with the fixes rather than the main neo. Are you comfortable enough with pip and git to do that yourself?

JohnStout, 2023-09-19:
Yup!

zm711, 2024-03-14:
@JohnStout 
Neo has updated the neuralynx .rawio. If you still need it we'd love to hear if the updates have worked for you.

zm711, 2024-06-19:
I'll close this for now. There has been a lot of work at the Neo level. If someone else runs into this in the future feel free to open an issue on Neo.

closed 2024-06-19 by zm711

JohnStout, 2024-06-19:
Sorry for not responding! Thank you! Haven’t used this interface in a bit
to know!