SpikeInterface/spikeinterface #2008: Request to Change Default Behavior of run_sorter() Function to Prevent Data Loss

DavidALloyd, 2023-09-18:
## Description:

Hello! I would like to request a change regarding the default behavior of the [spikeinterface.sorters.run_sorter()](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/sorters/runsorter.py) function.

As it currently stands, the [`run_sorter()`](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/sorters/runsorter.py#L94) function deletes everything in the output folder by default. While I understand that this may be intended for convenience, it has had a profoundly devastating impact on my work. I've been working with MEArec data and the sorting and extractor classes and wanted to dump the results into my output folder for this project. Unfortunately, this resulted in the immediate deletion of all other outputs, logs, results, and trained models I had carefully developed for this ongoing project, as well as my whole post-processed dataset. The loss has hit very hard, and and I want to make sure this doesn't happen to anyone else.

## Safety Concerns:
Deleting content by default can lead to unintended consequences, especially when working in production environments or with sensitive data. Users may expect the function to be non-destructive by default, and this behavior can easily lead to data loss or operational disruptions. Issues or mistakes with scope could also cause attempted deletion of root directories and entire drives!

The docs mention the parameter in two locations: the [sorters module description](https://spikeinterface.readthedocs.io/en/latest/modules/sorters.html) and the [API reference](https://spikeinterface.readthedocs.io/en/latest/api.html), but **neither** mention that the default behavior is to delete everything. The only clue is the function signature which shows that True for deleting the entire folder. I strongly recommend that there be a warning in the documentation if the default behavior for any function is data destruction for an entire directory.

## Suggested Improvement:
I kindly suggest changing the default parameter for remove_existing_folder to False, thereby requiring an explicit option to perform deletion. This approach aligns with best practices in library design, where destructive actions should require **explicit user consent**.

## Benefits:
* A significant reduction in the risk of accidental data loss.
* Enhanced user experience, providing users with the confidence that their data remains secure by default.
* Consistency with the principle of least surprise, where default behavior minimizes unexpected and harmful outcomes.

I believe that making this change will enhance the usability and safety of the library for all users. In the meantime I was able to recover one trained model, but everything else seems to be permanently gone. I will begin rebuilding from scratch as best I can.

Thank you for your attention to this request.

Environment:

    Python 3.11.2
    SpikeInterface 0.98.2

alejoe91, 2023-09-19:
Hi @DavidALloyd 

I'm sorry to hear about the data loss! Note that the default behavior is NOT to delete the output folder:
https://github.com/SpikeInterface/spikeinterface/blob/0.98.2/src/spikeinterface/sorters/runsorter.py#L95

Even when `delete_output_folder=True`, the function only deletes the files generated by the sorter. For example, if you set `output_folder="my_output_folder"`, all the files and folders generated by a sorter will be placed in the `my_output_folder/sorter_output` and only `sorter_output` will be deleted. 

Im not sure what happened in your case. It would be helpful to see the script that you used to check if the folder deletion could have happened somewhere else!

alejoe91, 2023-09-19:
I'm noticing now that you also refer to the `remove_existing_folder` argument. I totally agree this should be False by default and raise an error if the folder exists!

PR #2071 (2023-10-06): Change some default parameters for better user experience
Change some default parameters for better user experience
* waveform sparse by default
* ms_before=1.0,    ms_after=2.0,
* correlogram with bin_ms= 1.0
*  remove_existing_folder: bool = False for run_sorter()

close #2008

and also small fix (uncorrelated)
closed 2023-10-06 by alejoe91