SpikeInterface/spikeinterface #2010: y-coordinates fall outside of range. Glx, ks3, spikeinterface 0.98.1

Shrh627, 2023-09-19:
Hi,
I am using GLX and kilosort3, spikeinterface 0.98.1
The extractor is used like this: se.read_spikeglx( MYINPUT.ap.bin , stream_id = stream_id)
My parameters are:
"extractor_options": {

        "stream_id": "imec0.ap"

    },

    "sorter_options": {

        "detect_threshold": 6,

        "projection_threshold": [

            9,

            9

        ],

        "preclust_threshold": 8,

        "car": true,

        "minFR": 0.002,

        "minfr_goodchannels": 0,

        "nblocks": 5,

        "sig": 20,

        "freq_min": 300,

        "sigmaMask": 30,

        "nPCs": 3,

        "ntbuff": 64,

        "nfilt_factor": 4,

        "do_correction": true,

        "NT": 65600,

        "wave_length": 61,

        "keep_good_only": false,

        "n_jobs": 80,

        "chunk_duration": "1s",

        "progress_bar": true

    }
I encountered an issue with the channel map generation for a recording. I hope you can help me understand why this discrepancy occurred. I used the following criteria to select electrodes from shank 0:- Rows: 80 to 278- Y-coordinates: 1200 µm to 4080 µm However, when I examined the output channel map file, it displayed y-coordinates that fall outside of this range, specifically between 360 µm and 5145 µm. This is not in line with my selection criteria, and I'm curious if there are any options or configurations that need adjustment to correct this channel map. Could you please provide guidance on how to ensure that the channel map accurately reflects my electrode selection criteria? If there are any code snippets or configuration settings that would be helpful to review, please let me know. Thank you for your assistance.
Spikeinterface outpu
[image]
Correct channelmap
[image]

alejoe91, 2023-09-19:
Hi @Shrh627 

That's a Neuropixels 2.0 right?
Can you try to install the latest probeinterface? We fixed some NP2 mapping recently, whcih are going to be released soon!

You can just do the following:
```
pip install git+https://github.com/SpikeInterface/probeinterface.git
```
and run your code again

alejoe91, 2023-09-19:
Could you also share the `ap.meta` file in the SpikeGLX folder?

Shrh627, 2023-09-20:
Hi @alejoe91,

Yes, it's a NP2.0 recording. I'll try the installation and will run the code again. Thanks for your reply. Here is the meta file of the recording: 
[sh0br80bvd_g0_tcatmeta.txt](https://github.com/SpikeInterface/spikeinterface/files/12673150/sh0br80bvd_g0_tcatmeta.txt)

Shrh627, 2023-09-22:
Hi, with the probeinterface update the channelmap output looks correct. Thanks!

closed 2023-09-22 by alejoe91