SpikeInterface/spikeinterface #2017: Question about whitening - compute covariance matrix with or without spikes

JoeZiminski, 2023-09-19:
In some implementations ([KS](https://proceedings.neurips.cc/paper_files/paper/2016/file/1145a30ff80745b56fb0cecf65305017-Paper.pdf), Spyking Circus](https://www.biorxiv.org/content/10.1101/067843v1.full.pdf)) the spikes are removed before computing the covariance matrix used for whitening. In the [whiten.py pre-processing module](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/preprocessing/whiten.py) as far as I can tell the random chunks used for the whitening matrix computation contain spikes. Is it worth including an option to remove spikes using some absolute threshold?

It would be great to also here any thoughts / advice people have on the whitening step. My understanding it's primary use is to reduce the spatial extent of spikes across channels and to improve separation of clusters. In which case, I guess it would make sense to include the spikes in the estimation of the covariance matrix. Otherwise if the purpose is to remove correlated noise, then not including the spikes intuitively at least makes sense.

In general I am quite interested in this so if anyone can recommend any readings it would be much appreciated!

zm711, 2023-09-19:
I second this! I was trying to read up on whitening (in the neuroscience context) as well so super interested in others' insights! Thanks for bringing this up @JoeZiminski.

zm711, 2023-09-19:
I think that would be cool. I often get breathing artifacts across all channels so whitening based on that instead of with spikes might help.

alejoe91, 2023-09-19:
@zm711 this is an interesting one: https://link.springer.com/article/10.1007/s10827-015-0547-7

zm711, 2023-09-19:
Thanks @alejoe91 I'll give it a read.

Another question more related to @JoeZiminski's is that spikeinterface uses mean subtraction (optionally)
https://github.com/SpikeInterface/spikeinterface/blob/843b451282369565ca8e62863b4814c772c479f8/src/spikeinterface/preprocessing/whiten.py#L141C5-L141C7

But KS for example uses a mean subtraction, CAR, and median subtraction. So benefits of mean vs median in this case for preparing the data matrix pre-whitening.
https://github.com/MouseLand/Kilosort/blob/d55179f4bed45d4f17e5481283bc3f260212c1c7/preProcess/gpufilter.m#L22-L26

samuelgarcia, 2023-09-19:
@JoeZiminski
1. I am not sure that kilosort is removing spike for the covariance matrix.
   See https://github.com/MouseLand/Kilosort/blob/main/preProcess/get_whitening_matrix.m
2. Yes it would be better to remove spike for this covariance estimation.
3. whitenning is not a multichannel denoiser, it is more a noise decorelation
4. I prefer a robust zscore over whitenning in general because it use median+mad and so you do not car about spikes or 
    not in chunks

JoeZiminski, 2023-09-20:
Thanks everyone for your responses, that's very interesting / useful.

I can't either see in the KS code anywhere where spikes are removed or absolute thresholding on the buffer is done, that is surprising as the 2016 paper (and by implication the 2020 paper which references it in the 2016 paper in the whitening section) say this is how it was done. Maybe it is hidden in something subtle (though I really cannot think now) or removed after 2016.

I'd be happy to open a PR into adding a spike-free covariance matrix estimation option but am a bit swamped at the moment so will leave this open as a reminder.

That's interesting regarding robust zscore, so this is alternative to whitening (does it remove cross-channel correlated noise / do you have a reference?) At some point I'd love at some point to do a more thorough study on whitening and similar methods (e.g. [robust whitening](https://cmp.felk.cvut.cz/~chum/papers/Mukundan-SCIA17.pdf) and its effects on the data and spike sorting, would also be a good opportunity to learn MEARec, but of course the limiting factor is always time!

That's interesting @zm711, if I understand correctly I guess this depends on the preprocessing pipeline that is created in SI. For example if bandpass filtering is applied in the usual way the mean will be removed (along the time axis) anyway. I wonder how median subtraction across the channel axis will be affected by removal of the mean on the time axis, presumably it will have some effect as each channel is subtracted by a different scalar when removing the mean across the time axis. It's a good point about mean subtract prior to pre-whitening (this is talked about [here](http://ufldl.stanford.edu/tutorial/unsupervised/PCAWhitening/). I never think this matters as forget covariance is typically calculated with the dot product not the [other definition](https://www.google.com/search?sca_esv=566884281&rlz=1C1ONGR_en-GBGB990GB990&q=covariance+formula&tbm=isch&source=lnms&sa=X&sqi=2&ved=2ahUKEwjpvOqO-LiBAxW6SkEAHZrCASQQ0pQJegQIDRAB&biw=2133&bih=1070&dpr=0.9#imgrc=70rtEwYlc_B2jM that includes mean-subtraction.

zm711, 2023-09-20:
@JoeZiminski 

Did you also look (and not to give you more work) in the KS1 repo? https://github.com/cortex-lab/KiloSort to see if that is what they're 2016 paper is referring to?

JoeZiminski, 2023-09-20:
Thanks @zm711! Yes it seems a [no-spikes version](https://github.com/cortex-lab/KiloSort/blob/cd040da1963dd760da98b54c811b3fd441d54e79/preprocessData.m#L149C1-L149C1) was removed, the `otherwise` block is what made it into KS2.